bishoph/sopare #1: Basic usage?

magdesign, 2017-04-13:
Pretty cool tool you developed!

I installed on my Rpi3 and can run the sopare.py
But I guess I first have to do something like learning words?
Can you please describe in a few basic words how the workflow of this tool goes.
Thank you very much!

btw: searching a simple voice control tool to make an art installation where you can ask the figures who they are.... messed around with jasper and wit.ai, but in the gallery I am offline....

bishoph, 2017-04-13:
First, thanks that you like the project. Second, I plan a short basic tutorial that should be available next week. I got already some request on bishoph.org.

Anyway, I can give you a very brief overview about how to start. 

Sopare must be trained to get results, right. And to do this, make sure to configure the mic threshold. This can be done in the file sopare/config.py with the parameter "THRESHOLD". To figure out your threshold start sopare in verbose mode with the command

./sopare.py -l -v

You should not see any debug output until you start talking. Increase/decrease the threshold until your environment noise does not affect your voice input.

Next you train your words. Simply start sopare like this:

./sopare.py -t YOUR_WORD_IDENTIFIER

Start speaking your word after the line "start endless recording" appears on the screen. Repeat to train one word 3-5 times. After your training is complete, run the command 

./sopare.py -c

to compile the training data. Now you can start sopare and test your trained input by running the command

./sopare.py -l

If sopare recognize your trained sound input the YOUR_WORD_IDENTIFIER gets printed on the screen. There are many config option to adjust the accuracy, but you should get some decent results with the basic settings...

magdesign, 2017-04-14:
Thanks for your short explanation!
It worked so far. First I had to use another USB soundcard since the one I used had a pretty bad recording quality. For other people reading this, you have to set the USB soundcard as default.
On debian jessie, open: `sudo nano /usr/share/alsa/alsa.conf`
and scroll down to 
```
defaults.ctl.card 0
defaults.pcm.card 0
```
change the 0 to a 1, then reboot.

So i start to training, I guess i should also use female voices for the same words to make it more accurate...

magdesign, 2017-04-14:
The tool works fine in the first usage.
But when I kill it with CTRL+C and want to train it some more words like: ./sopare.py -t WIESO

it crashes with an error:
```
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused
stream read error [Errno Input overflowed] -9981
```

After a reboot it works fine again. Think that the alsa input is still in use...

magdesign, 2017-04-14:
Update

I got rid of the errors by removing following entries from /usr/share/alsa/alsa.conf:

```
pcm.rear cards.pcm.rear
pcm.center_lfe cards.pcm.center_lfe
pcm.side cards.pcm.side
pcm.hdmi cards.pcm.hdmi
pcm.modem cards.pcm.modem
pcm.phoneline cards.pcm.phoneline

```

and removing all files having pulse in the name from: `/usr/share/alsa/alsa.conf.d/`

bishoph, 2017-04-20:
As you pointed out the ALSA information messages can be eliminated by cleaning up the "/usr/share/alsa/alsa.conf" file. This can be done by commenting out lines that are not in use or not available.

The "stream read error -9981" is a PyAudio warning that the sound buffer was not read fast enough. As sopare reads the input and hand it over to another process on a separate core there is not much I can do programmatically. As the buffer is already very small (parameter CHUNK in config.py) I could ignore the error with the PyAudio setting "exception_on_overflow=True"...but as the impact is not really noticeable I would like to keep it as it is at the moment.

bishoph, 2017-05-11:
As the basic tutorial is available I close this issue:
http://www.bishoph.org/sopare-basic-usage-voice-controlling-a-magic-mirror/

closed 2017-05-11 by bishoph