bishoph/sopare #2: Take pattens from files generated in .wav / .raw format

Dakicha, 2017-05-10:
Adrian says:
8. Mai 2017 at 20:16
Thank you very much for sharing this project.
It is very interesting and works without any effort on Orange PI LITE.

Question: Can take the pattens from files already generated in .wav format?
This would avoid having to use the microphone to capture each of the sounds.

Greetings from Mallorca.

Antworten
bishoph says:
8. Mai 2017 at 22:16
You are welcome and great to hear that it works for you on an Orange PI LITE!

In regards to your questions: training from „wav“ files is currently not supported. But there is an option to store recorded input:
../sopare.py -w samples/test.raw
and the recorded file can be used for training or testing:
./sopare.py -r samples/test.raw – t test -v

As this options are already available, it should be possible to create a converter to transform „wav“ to „raw“ files, to achieve what you are asking for. I’ve added this to my to-do list, but I remember something that there was an issue converting „wav“ files so it’s not guaranteed that this becomes a feature 😉

Saludos

Antworten
Adrian says:
9. Mai 2017 at 10:21
Thank you for answering so quickly!

With your suggestion I have found this great tutorial where it explains how to do the conversion
https://www.hellomico.com/getting-started/convert-audio-to-raw/

After that, I have using:
./sopare.py -r samples/test.raw -t test -v
./sopare.py -c
But there are no changes in the dict.json file so it does not recognize the new „test“ pattern.
There is any way to attach or send the test.raw so that you take a look and validity this method of conversion?

Thanks in advance!

Antworten
Adrian says:
9. Mai 2017 at 10:55
I have also tried:

./sopare.py -w samples/test.raw
./sopare.py -r samples/test.raw -t test -v
./sopare.py -c
The same, no changes.

But, if you open the test.raw file generated by sopare in audacity, the format is Signed 16 bit PCM – Little-endian – 1 Channel (MONO) and the audio is correct, same like generated by the conversion from wav to raw.

Antworten
bishoph says:
9. Mai 2017 at 18:02
Hmm, what’s possible is that the volume is below the THRESHOLD and therefore „test“ doesn’t show up in the dict.json file as it was never learned, but this is just a wild guess. I tested again and as expected, everything works in my environment 🙂

I would like to suggest to move this thread and the issue we are discussing to GitHub, as you can easily attach files, track the issue and I can close it when its done: https://github.com/bishoph/sopare/issues/new

BTW: You can check what’s already learned with the command:

./sopare.py -o

The output is a string pair which consists of the ID and a UUID, which corresponds with the a JSON file in the dict/directory (even if the suffix pretends something different) …

In the case you want to see only unique learned IDs, this command chain is quite handy:

./sopare.py -s '*' | sed 's/[^a-z].*//' | sed 's/\//g' | grep -v '^$' | sort | uniq
Thanks y un saludo 🙂

Dakicha, 2017-05-10:
Hola!

It did not work because a silence of more than 3 seconds is needed at the end of the audio (pattern) inside the wav / raw. Just I add 5 seconds of silence after each pattern and now the option:
./sopare.py -r samples/test.raw -t test -v 
work and generate the dict file correctly after:
./sopare.py -c

But the next problem is that this training does not work for my.
Only "work" if I training the song with normal usage by capturing thru the mic.
I said "work" because I have too much false positives. (The next step will be to adjust the accuracy!)
This is why I want to capture the pattern directly from a wav file, the accuracy!.

In my case, I have an old machine that says the states in which is mediating sounds of numbers (0, 1, 2, 3,...). These number already have stored in wav and for this reason I want to use the audio line output of this machine  to the line in of OrangePi Lite, In this way I only need to detect the same patterns every time. Without significant variations of sound.

Any suggestions on how to do it?

Muchas gracias amigo, y si alguna vez vienes a Mallorca avisame que invito las cervezas!

bishoph, 2017-05-11:
The issue, that the raw can't be used for training, is a bug that I'm going to fix whenever I have some time. Thanks for finding and reporting the issue!

In regards to the accuracy: The default config options are optimize for human voice and depending on your frequency domain, you maybe have to tweak a lot of settings to increase precision. As you compare later on through the mic, sopare must compare the sound including background noise, echos and other stuff that may interfere and causes deviations. That said, I don't think that a converted "wav" sample gets much better accuracy than a clean trained pattern through the mic. On the other hand if you train and compare generated "wav/raw" files from a clean and well defined input source, then you should be able to get real precise results and less to no false positives. Hope that helps to manage your expectations and to get you going. 

Perfecto, y muy similares :)

bishoph, 2017-05-20:
This issue should be fixed in the testing branch: 

https://github.com/bishoph/sopare/commit/faafdd097a5e3a96308a5b004f582b4f2e135f7f

Now I need some time to make a tutorial about increasing precision and accuracy... :)

bishoph, 2017-09-24:
Fixed with stable version >= 1.3.0 and the tutorial about increasing precision and accuracy is available. Closing issue.

closed 2017-09-24 by bishoph