bishoph/sopare #3: Noise Filtering?

ankit-gaur, 2017-10-10:
First of all, thanks for developing this.
I want to ask that have you integrated something for noise filtering/cancellation from mic stream/wav file? I tried to look into the code but couldn't find.

bishoph, 2017-10-14:
There is no specific noise cancellation. But SOPARE uses a configurable range of frequencies for the analysis.  Furthermore, SOPARE compresses the frequencies and creates a characteristic. And you have options to define the accuracy which helps to identify true positives with background noise or in different conditions.

Maybe you find this helpful: https://www.bishoph.org/sopare-precision-and-accuracy/

bishoph, 2018-02-07:
Closing issue

closed 2018-02-07 by bishoph