bishoph/sopare #4: stream read error [Errno -9988] Stream closed

GordonTraveler, 2017-11-06:
Hi Bishoph.

Your project sounds really nice and probably exactly what I am looking for... a lightweight speech recognition for the RPi. Unfortunately it is not working.
I use a RPi 2 with the new Raspbian Stretch lite. I installed all dependencies from your readme.
The software is going to start:

`pi@RPI2:~/sopare $ sudo python sopare.py -l >> run.log
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
`

The output of the run.log is:

`sopare 1.3.0
start endless recording
sopare 1.3.0
start endless recording
stream read error [Errno -9981] Input overflowed
stream read error [Errno -9988] Stream closed
stream read error [Errno -9988] Stream closed
... million of times
`

I guess it is an issue with the sound system. I use a USB Soundcard, which is device 0. Internal is device 1.

How to solve this problem? I guess I need to tell the recorder that the mix is at hwplug:0,0.

bishoph, 2017-11-06:
Hi,

seems that I really looked ahead by adding an audio test case with recommendation with the last commit ;)
That said, please checkout the latest testing branch and run the audio tests with at least sopare version 1.3.7:
python test/test_audio.py

In all of my test cases (except for one) changing the sample rate eliminates the "Errno -9988" error messages from pyaudio. The one exception was a poor and very cheap USB mic that still lead to "Errno -9981" errors, but you can ignore these kind of error messages as it is more a warning.

Happy voice controlling :)

GordonTraveler, 2017-11-06:
Ahh, ok. Now I found the testing branch :-) 
I checked it out and run the test_audio.py

> pi@RPI2:~/sopare-test $ python test/test_audio.py
> test_audio init...
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
> Cannot connect to server socket err = No such file or directory
> Cannot connect to server request channel
> jack server is not running or cannot be started
> JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
> JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
> 
> 
> ##### Default input device info #####
> defaultSampleRate: 44100.0
> defaultLowOutputLatency: 0.0087074829932
> defaultLowInputLatency: 0.00868480725624
> maxInputChannels: 128
> structVersion: 2
> hostApi: 0
> index: 11
> defaultHighOutputLatency: 0.0348299319728
> maxOutputChannels: 128
> name: default
> defaultHighInputLatency: 0.0348299319728
> #####################################
> 
> 
> testing different SAMPLE_RATEs ... this may take a while!
> 
> 
> testing different CHUNK sizes ... this may take a while!
> 
> 
> Excellent. Got all 5000 chunks.
> Excellent. Got all 10000 chunks.
> Excellent. Got all 20000 chunks.
> Excellent. Got all 40000 chunks.
> Excellent. Got all 80000 chunks.
> Excellent. Got all 5000 chunks.
> Excellent. Got all 10000 chunks.
> Excellent. Got all 20000 chunks.
> Excellent. Got all 40000 chunks.
> Excellent. Got all 80000 chunks.
> Excellent. Got all 5000 chunks.
> Excellent. Got all 10000 chunks.
> Excellent. Got all 20000 chunks.
> Excellent. Got all 40000 chunks.
> Excellent. Got all 80000 chunks.
> Excellent. Got all 5000 chunks.
> Excellent. Got all 10000 chunks.
> Excellent. Got all 20000 chunks.
> Excellent. Got all 40000 chunks.
> Excellent. Got all 80000 chunks.
> Excellent. Got all 5000 chunks.
> Excellent. Got all 10000 chunks.
> Excellent. Got all 20000 chunks.
> Excellent. Got all 40000 chunks.
> Excellent. Got all 80000 chunks.
> Excellent. Got all 5000 chunks.
> Excellent. Got all 10000 chunks.
> Excellent. Got all 20000 chunks.
> Excellent. Got all 40000 chunks.
> Excellent. Got all 80000 chunks.
> Error: [Errno -9981] Input overflowed
> Error: [Errno -9981] Input overflowed
> Error: [Errno -9981] Input overflowed
> Error: [Errno -9981] Input overflowed
> Error: [Errno -9981] Input overflowed
> Excellent. Got all 5000 chunks.
> Excellent. Got all 10000 chunks.
> Excellent. Got all 20000 chunks.
> Excellent. Got all 40000 chunks.
> Excellent. Got all 80000 chunks.
> 
> 
> 
> Your sopare/config.py recommendations:
> 
> SAMPLE_RATE = 48000
> CHUNK = 500
> THRESHOLD = 3200

I created a config.py and added the recommendations. No changes. Same issue.

I tried:
> arecord -f S16_LE -r 48000 -D hw:0,0 -d 10 ~/testS16_LE.wav
and this is working fine. So the sample rate should be ok, isn't? 

I am not the Linux expert. Don't know what I can try.

bishoph, 2017-11-06:
As hw:0,0 points already to the default your alsa configuration seems to be fine. If you still get the same errors my only guess is that you have not changed the config.py file SOPARE reads. Let us go step by step:

You need to change the option "SAMPLE_RATE " in the existing config file "./sopare/config.py" and change it from the default value of "44100" to the recommended value of "48000". Save the existing file and make sure the value changed (e.g. with the command "git diff sopare/config.py"). Now stop all SOPARE instances that are eventually running and try again. The error "Errno -9988" should be gone as the audio test was successful with this value and arecord works also with this value - as you've said.

GordonTraveler, 2017-11-06:
Ok, I found the config file now :-) Changed the entries, but still the same. 

> pi@RPI2:~/sopare-test $ sudo python sopare.py -l -v
> sopare 1.3.8
> ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
> ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
> Cannot connect to server socket err = No such file or directory
> Cannot connect to server request channel
> jack server is not running or cannot be started
> JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
> JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
> No handlers could be found for logger "sopare.recorder"
> INFO:root:checking for plugins...
> DEBUG:root:loading and initialzing plugins/print
> INFO:sopare.recorder:start endless recording
> INFO:sopare.worker:worker queue runner started
> INFO:sopare.buffering:buffering queue runner
> WARNING:sopare.recorder:stream read error [Errno -9981] Input overflowed
> WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
> WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
> WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
> WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
> WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
> WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
> WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
> WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
> WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
> ....

The config file:

> #########################################################
> # Stream prep and silence configuration options #########
> #########################################################
> 
> # Read chunk size
> CHUNK = 500
> 
> # Sample rate
> SAMPLE_RATE = 44800
> 
> # Volume threshold when audio processing starts / silence
> THRESHOLD = 3200
> 
> # Silence time in seconds when analysis is called
> MAX_SILENCE_AFTER_START = 1
> 
> # Time in seconds after the analysis is forced
> MAX_TIME = 2.4
> 
> # Start the analysis after reaching LONG_SILENCE
> LONG_SILENCE = 30
> 
> # Characteristic length
> CHUNKS = 1024*3

I tried also the sample rate of 44100. But the same.

My usb soundcard is:
> Bus 001 Device 007: ID 0d8c:0014 C-Media Electronics, Inc.

If I can provide any other information, let me know how.

bishoph, 2017-11-06:
The recommended sample rate is 48000 and not 44800 :)

pingus69, 2017-11-06:
Hi,

first: sopare is exactly what I was looking for. It looks great. Thanks for you work!!
I want to control a model railway via speech.
But unfortunately I have the same problems with pyaudio. I always get 

sopare 1.3.0
start endless recording
stream read error [Errno -9981] Input overflowed
stream read error [Errno -9988] Stream closed
stream read error [Errno -9988] Stream closed
stream read error [Errno -9988] Stream closed
stream read error [Errno -9988] Stream closed
stream read error [Errno -9988] Stream closed
stream read error [Errno -9988] Stream closed
stream read error [Errno -9988] Stream closed
stream read error [Errno -9988] Stream closed
stream read error [Errno -9988] Stream closed

Thanks

Bernward

pingus69, 2017-11-06:
I am using Raspberry Pi Model B with Jessie and 
USB Mikrofon "AKIRO" Kinobo. At the beginning I get 
xcb_connection_has_error() returned true
ALSA lib pcm_dmix.c:1022:(snd_pcm_dmix_open) unable to open slave
xcb_connection_has_error() returned true
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

xcb_connection_has_error() returned true
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

ALSA lib pcm_dmix.c:1022:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm_dmix.c:1022:(snd_pcm_dmix_open) unable to open slave
connect(2) call to /tmp/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed

Thanks

Bernward

bishoph, 2017-11-06:
Hi Bernward,

in regards to the "Errno -9988" error message, could you please also checkout the testing branch and run the audio tests like described above. In my experience you can avoid this by choosing a different sample rate and the file "test/test_audio.py" helps you to find the best values.

In terms of the second post, as far as I can tell to avoid the messages you need to clean your "/usr/share/alsa/alsa.conf" config file. The messages are all from the underlying sound config and are not SOPARE related ....

Have fun and happy voice controlling :)

pingus69, 2017-11-06:
I played the raw-Files under dict. Only sough

bishoph, 2017-11-06:
The above messages means that the multiprocessor launched close/stop methods are running while the main program stops. This means normally that your timeout values are too low. Try something some higher values like

MAX_SILENCE_AFTER_START = 2
MAX_TIME = 3

In term of recognition. Like stated in the section "Quick start, useful commands and usage examples" here (https://github.com/bishoph/sopare/tree/testing): Changing config options ... requires re-training. Delete your training files and the dictionary entries, then train again. Follow the steps of my blog post https://www.bishoph.org/sopare-precision-and-accuracy/ to start rough to get lots of results and then add precision to avoid false positives and get better results.

pingus69, 2017-11-06:
With sopare -t <word> 
i always get a

Buffering not alive, stop recording

bishoph, 2017-11-07:
Yeah, that's normal in version 1.3.0 as the recording stops after the configured time. In the testing branch version you don't get this message as I've already overworked the logging output :)

GordonTraveler, 2017-11-07:
So, I tried several Sample Rates. With 16.000 the error disappears. The output is now:

> pi@RPI2:~/sopare-test $ python sopare.py -v -l
> sopare 1.3.8
> ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
> ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
> ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
> Cannot connect to server socket err = No such file or directory
> Cannot connect to server request channel
> jack server is not running or cannot be started
> JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
> JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
> No handlers could be found for logger "sopare.recorder"
> INFO:root:checking for plugins...
> DEBUG:root:loading and initialzing plugins/print
> INFO:sopare.recorder:start endless recording
> INFO:sopare.worker:worker queue runner started
> INFO:sopare.buffering:buffering queue runner
> INFO:sopare.processing:starting append mode
> INFO:sopare.processing:stop append mode because of silence
> DEBUG:sopare.filter:New window!
> /home/pi/sopare-test/sopare/filter.py:133: FutureWarning: comparison to `None` will result in an elementwise object comparison in the future.
>   if (shift_fft != None and (hasattr(sopare.config, 'FFT_SHIFT') and sopare.config.FFT_SHIFT == True)):
> INFO:sopare.buffering:stop buffering
> DEBUG:sopare.worker:characteristic = 0 {'dfm': 7126449, 'peaks': [], 'df': 125, 'volume': 0, 'fc': 211.0, 'norm': [0.09253204612828189, 0.09487714472512919, 0.10620974729869678, 0.1116215183340881, 0.12211180080237569, 0.11723676187862372, 0.11010453444071039, 0.10666234597927678, 0.10925479991432518, 0.097799072342522, 0.12464225800385019, 0.10452192257268646, 0.11206989239876415, 0.10965130648543706, 0.11392860536823822, 0.10891454381574849, 0.11441621941252604, 0.11189442524355613, 0.11563560612950216, 0.12592963592566714, 0.13361292340461217, 0.13583567616886616, 0.13434836729059646, 0.14171272527443482, 0.13525759659830358, 0.1387096543910757, 0.1384734204743991, 0.15222872167322796, 0.15221236398757435, 0.1402126958062954, 0.13404219766676614, 0.13223379593088316, 0.13770590292237567, 0.1470090785128768, 0.1547288932345064, 0.15463727622651408, 0.14969958118588303, 0.15291613239874108, 0.15186466747027236, 0.15205498840478296, 0.1487678151957922, 0.1468342605799082, 0.1411653215741634, 0.13823353344288292, 0.13815272542593876, 0.13948951774153023, 0.13882539004536198, 0.13837698642347343, 0.14462513804963373, 0.14197036867472407, 0.1349054097185554, 0.13976695180950896, 0.13751778697461406, 0.10926271637546837, 0.13462992725560163, 0.1425283174768301, 0.1294524413641999, 0.12946414610479204], 'token_peaks': []}
> DEBUG:sopare.worker:meta = [{'adapting': 0, 'peaks': [2656727, 2357712, 2928906, 3241243, 2717335, 2626076, 2953984, 2671660], 'pos': 8, 'volume': 0, 'token': 'start analysis', 'silence': 4}]
> DEBUG:root:************************************************
> 
> [({'dfm': 7126449, 'peaks': [], 'df': 125, 'volume': 0, 'fc': 211.0, 'norm': [0.09253204612828189, 0.09487714472512919, 0.10620974729869678, 0.1116215183340881, 0.12211180080237569, 0.11723676187862372, 0.11010453444071039, 0.10666234597927678, 0.10925479991432518, 0.097799072342522, 0.12464225800385019, 0.10452192257268646, 0.11206989239876415, 0.10965130648543706, 0.11392860536823822, 0.10891454381574849, 0.11441621941252604, 0.11189442524355613, 0.11563560612950216, 0.12592963592566714, 0.13361292340461217, 0.13583567616886616, 0.13434836729059646, 0.14171272527443482, 0.13525759659830358, 0.1387096543910757, 0.1384734204743991, 0.15222872167322796, 0.15221236398757435, 0.1402126958062954, 0.13404219766676614, 0.13223379593088316, 0.13770590292237567, 0.1470090785128768, 0.1547288932345064, 0.15463727622651408, 0.14969958118588303, 0.15291613239874108, 0.15186466747027236, 0.15205498840478296, 0.1487678151957922, 0.1468342605799082, 0.1411653215741634, 0.13823353344288292, 0.13815272542593876, 0.13948951774153023, 0.13882539004536198, 0.13837698642347343, 0.14462513804963373, 0.14197036867472407, 0.1349054097185554, 0.13976695180950896, 0.13751778697461406, 0.10926271637546837, 0.13462992725560163, 0.1425283174768301, 0.1294524413641999, 0.12946414610479204], 'token_peaks': []}, [{'adapting': 0, 'peaks': [2656727, 2357712, 2928906, 3241243, 2717335, 2626076, 2953984, 2671660], 'pos': 8, 'volume': 0, 'token': 'start analysis', 'silence': 4}])]
> 
> {}
> 
> framing_match: []
> 
> sorted_best_match: []
> 
> match_results: ['']
> 
> Results contain too many empty tokens. 1 / 1 Eliminating results
> []
> INFO:sopare.processing:starting append mode
> INFO:sopare.processing:stop append mode because of silence
> INFO:sopare.buffering:stop buffering
[... cut ...] 277542314525419, 0.12548001835718048, 0.13208252930639144, 0.13410813871274224], 'token_peaks': []}, [{'adapting': 0, 'peaks': [2767715, 2976046, 2710046, 3268586, 2722398, 2603943, 2523570, 2555145], 'pos': 8, 'volume': 0, 'token': 'start analysis', 'silence': 4}])]
> 
> {}
> 
> framing_match: []
> 
> sorted_best_match: []
> 
> match_results: ['']

I guess that is a more normal output. I'll try to build a dictionary now and if I don't come back... then it was working... if not, I'll be back :-)

bishoph, 2017-11-21:
Please note that I had a system running Raspbian Jessie just fine with a sample rate of 44100. After an update to Raspbian Stretch I also got the message "stream read error [Errno -9988] Stream closed" all over the place and was no longer able to train or run SOPARE properly. I changed the sample rate to 8000 with a CHUNK value of 512 and was able to train and run SOPARE normally. This said, the test/test_audio.py is running fine on both versions with the sample rate of 44100. As SOPARE did not change, the underlying (Python) stack changed and something is running much slower. I have to dig into this but as SOPARE can be configured to run normally so I close this issue.

closed 2017-11-21 by bishoph

bishoph, 2017-11-21:
Ok, I tested a lot and hopefully/finally I fixed the issue. On all of my systems the message "stream read error [Errno -9988] Stream closed" is gone while using the 

python test/test_audio.py

recommendation. Please reopen if somebody encounter the message with a SOPARE version >= 1.4.2 (currently in the testing branch) while using the recommendations from the test_audio test case!
Ok, I tested a lot and hopefully/finally I fixed the issue. On all of my systems the message "stream read error [Errno -9988] Stream closed" is gone while using the 

python test/test_audio.py

recommendation. Please reopen if somebody encounter the message with a SOPARE version >= 1.4.2 (currently in the testing branch) while using the recommendations from the test_audio test case!

arnold-b, 2018-11-25:
Hello, 
sorry that i need to tell you i also have this error. I installed your actual master and i also tested your testing version. on both i cant train the system.  
I also tried a few diffrent microphone but always with the same usb / audio
raspberrypi:~/dev/sopare# python --version
Python 2.7.13
root@raspberrypi:~/dev/sopare# python sopare.py -u
sopare 1.5.3
starting unit tests...
starting analyze tests...
analyze test preparation...
testing analyze get_match...
testing normal conditions (1)[u'test1'] == [u'test1']
testing normal conditions (2)[u'test1', u'test3'] == [u'test1', u'test3']
testing normal conditions (3)[u'test1', u'test3', u'test2'] == [u'test1', u'test3', u'test2']
testing leading space [u'test1', u'test3', u'test2'] == [u'test1', u'test3', u'test2']
testing ending space [u'test1', u'test3', u'test2'] == [u'test1', u'test3', u'test2']
testing correct order [u'test1', u'test3', u'test2', u'test1', u'test3', u'test2'] == [u'test1', u'test3', u'test2', u'test1', u'test3', u'test2']
testing strict length [u'test1', u'test3', '', u'test2'] == [u'test1', u'test3', '', u'test2']
testing false leading results ['', u'test1', '', u'test2'] == ['', u'test1', '', u'test2']
analyze tests run successful.
filter test preparation...
testing filter n_shift...
testing n_shift [5, 6, 7, 8, 9, 10, 11, 12, 13, 14] == [5, 6, 7, 8, 9, 10, 11, 12, 13, 14]
testing n_shift [15, 16, 17, 18, 19, 20, 21, 22, 23, 24] == [15, 16, 17, 18, 19, 20, 21, 22, 23, 24]
testing n_shift [25, 26, 27, 28, 29, 30, 31, 32, 33, 34] == [25, 26, 27, 28, 29, 30, 31, 32, 33, 34]
filter tests run successful.
unit_tests run successful!
done.

WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
[... 2065 lines cut ...]
    res = self._recv()
    KeyboardInterrupt
recorder.recorder(cfg)
  File "/root/dev/sopare/sopare/recorder.py", line 44, in __init__
    res = self._recv()
KeyboardInterrupt
    self.recording()
  File "/root/dev/sopare/sopare/recorder.py", line 95, in recording
    self.logger.warning("stream read error "+str(e))
  File "/usr/lib/python2.7/logging/__init__.py", line 1179, in warning
    self._log(WARNING, msg, args, **kwargs)
  File "/usr/lib/python2.7/logging/__init__.py", line 1285, in _log
    record = self.makeRecord(self.name, level, fn, lno, msg, args, exc_info, func, extra)
  File "/usr/lib/python2.7/logging/__init__.py", line 1259, in makeRecord
    rv = LogRecord(name, level, fn, lno, msg, args, exc_info, func)
  File "/usr/lib/python2.7/logging/__init__.py", line 267, in __init__
    self.levelname = getLevelName(level)
  File "/usr/lib/python2.7/logging/__init__.py", line 162, in getLevelName
    return _levelNames.get(level, ("Level %s" % level))
KeyboardInterrupt
root@raspberrypi:~/dev/sopare#

i hope you can help ?

thanks Arnold

arnold-b, 2018-11-25:
audio device: Bus 001 Device 004: ID 0ccd:0077 TerraTec Electronic GmbH Aureon Dual USB

bishoph, 2018-11-25:
Hi Arnold, There are several samples rates that seems to work. Try some more combinations like 44100 or 16000 with both, 512 and 1024 CHUNKS beside the 48000 recommendation. Some mics needs very specific combos to work without warnings.
One hint: a THRESHOLD of 17500 seems to be too high and will probably not produce good results (if any). You should run the test in a silent environment as written in the documentation. As I don't know about your goals and the environment I can only say that my THRESHOLD values are normally between 100 and 800 for speech recognition in different environments and for different reasons. See my blog post  for more information.