bishoph/sopare #5: Always "Buffering not alive, stop recording" when using sopare -t <word>

pingus69, 2017-11-06:
Hi,

when I Use

sopare -t <word>

I get always the message:

start endless recording
Buffering not alive, stop recording
stop endless recording

or a

start endless recording
Process buffering queue:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/home/pi/sopare/sopare/buffering.py", line 42, in run
    self.proc.check_silence(buf)
  File "/home/pi/sopare/sopare/processing.py", line 82, in check_silence
    self.stop("stop append mode because of silence")
  File "/home/pi/sopare/sopare/processing.py", line 51, in stop
    self.prepare.stop()
  File "/home/pi/sopare/sopare/prepare.py", line 72, in stop
    self.filter_reset()
  File "/home/pi/sopare/sopare/prepare.py", line 89, in filter_reset
    self.filter.reset()
  File "/home/pi/sopare/sopare/filter.py", line 42, in reset
    self.queue.put({ 'action': 'reset' })
  File "/usr/lib/python2.7/multiprocessing/queues.py", line 100, in put
    assert not self._closed
AssertionError
Buffering not alive, stop recording

if I play a .raw-file in the dict-directory with

aplay <raw-file>

I cannot hear any of my spoken words. I hear only random noise.

I am Using Raspberry Pi 3 Model B with Jessie and a USB Kinobo microphone.

Here the first lines of my config.py:

CHUNK = 512

# Sample rate
#SAMPLE_RATE = 44100
SAMPLE_RATE = 32000

# Volume threshold when audio processing starts / silence
THRESHOLD = 250

# Silence time in seconds when analysis is called
MAX_SILENCE_AFTER_START = 1

# Time in seconds after the analysis is forced
MAX_TIME = 3.2

# Start the analysis after reaching LONG_SILENCE
LONG_SILENCE = 30

# Characteristic length
CHUNKS = 2048

I tried it also with CHUNK = 500. Does also not work.

bishoph, 2017-11-07:
The output above is normal for version 1.3.0 and below as it is the normal program information output. In the testing branch I've overworked the logging and the messages only show up in certain log levels.

Please note that dict/*.raw files are NOT raw audio files but a JSON structure. If you want to record raw audio please use the "-w --write" options like written in the help/usage instructions:

"./sopare.py -w samples/test.raw"

To avoid the multiprocessing error/warning messages it should be sufficient to increase the "MAX_SILENCE_AFTER_START" option to a higher value. When I have lots of time I'll look after that issue.

pingus69, 2017-11-07:
I played a little bit with the THRESHOLD. Seems to work now better. I will do more testing.
Thanks a lot.

bishoph, 2017-11-07:
Please try the "-~" respectively the "--wave" option. With this option SOPARE creates wave files for each token in the "token/" directory. Tokens are used to train and identify the sound. If you don't get some good token results you need to tweak around until you get good results, otherwise SOPARE can't be trained or recognize the sounds. 

For example: If I say the word "light" I get 3 tokens and I can hear something like "li-gh-t" if I play all three tokens. 

As I said before: Increase the MAX_SILENCE_AFTER_START  value to "3" and maybe the LONG_SILENCE to an absurd high value like "200". You can decrease later on after you hear your sound. In addition I wouldn't filter frequencies until you know whats going on. This means adjust the config to something like:

LOW_FREQ = 0
HIGH_FREQ = 5000

After that, if you can hear your sound you can write, read and train the recorded raw sound to test if the sound gets recognized:

./sopare.py -w samples/test.raw
./sopare.py -r samples/test.raw -t test
./sopare.py -c
./sopare.py -r samples/test.raw 

Now you must get ['test'] as expected and you can continue adjusting for better/faster results.

bishoph, 2017-11-21:
Fixed with SOPARE version >= 1.4.1, currently in testing branch.

closed 2017-11-21 by bishoph