bishoph/sopare #7: Startup error

hellowtisch, 2017-11-19:
Hello, 
when trying to run the test train script, I receive the following error.
I use a fresh RPi3 with latest Raspbian stretch on it.

 _```
./sopare.py -t -v "Test"
sopare 1.4.0
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.front.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
/home/pi/sopare/sopare/filter.py:133: FutureWarning: comparison to `None` will result in an elementwise object comparison in the future.
  if (shift_fft != None and (hasattr(sopare.config, 'FFT_SHIFT') and sopare.config.FFT_SHIFT == True)):
Process buffering queue:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/home/pi/sopare/sopare/buffering.py", line 44, in run
    self.proc.check_silence(buf)
  File "/home/pi/sopare/sopare/processing.py", line 83, in check_silence
    self.stop("stop append mode because time is up")
  File "/home/pi/sopare/sopare/processing.py", line 52, in stop
    self.prepare.stop()
  File "/home/pi/sopare/sopare/prepare.py", line 72, in stop
    self.filter_reset()
  File "/home/pi/sopare/sopare/prepare.py", line 89, in filter_reset
    self.filter.reset()
  File "/home/pi/sopare/sopare/filter.py", line 48, in reset
    self.queue.put({ 'action': 'reset' })
  File "/usr/lib/python2.7/multiprocessing/queues.py", line 100, in put
    assert not self._closed
AssertionError
```_

Do you have any ideas on how to fix this?

bishoph, 2017-11-20:
SOPARE uses multiple instances across multiple CPU cores. The issue above shows that one instance is still running while the main instance shuts down. I have already a wait loop implemented but it seems that the given time is not being enough.

Please try to increase the value (.1) in the file sopare/recorder.py (line 91):
time.sleep(.1) # wait for all threads to finish their work

Would be perfect if you could tell me if the issue can be fixed by increasing the above value and if yes, the value when the issue disappears...

hellowtisch, 2017-11-20:
Hi Martin,

 

unfortunately this did not seem to change a lot.

I changed the value from .1. to .5 and then to 1 with the following result:

 

pi@raspberrypi:~/sopare $ ./sopare.py -t "test"

sopare 1.4.0

ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.front.0:CARD=0'

ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory

ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side

ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'

ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory

ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21

ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'

ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory

ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21

ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround40.0:CARD=0'

ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory

ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround40

ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'

ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory

ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround41

ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'

ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory

ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround50

ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'

[... 92 lines cut ...]
  File "/home/pi/sopare/sopare/processing.py", line 83, in check_silence

    self.stop("stop append mode because time is up")

  File "/home/pi/sopare/sopare/processing.py", line 52, in stop

    self.prepare.stop()

  File "/home/pi/sopare/sopare/prepare.py", line 72, in stop

    self.filter_reset()

  File "/home/pi/sopare/sopare/prepare.py", line 89, in filter_reset

    self.filter.reset()

  File "/home/pi/sopare/sopare/filter.py", line 48, in reset

    self.queue.put({ 'action': 'reset' })

  File "/usr/lib/python2.7/multiprocessing/queues.py", line 100, in put

    assert not self._closed

AssertionError

bishoph, 2017-11-20:
Ups, sorry. Reading the docs and the code helps. My fault. I'm trying to fix this with the next update in the testing branch - but this may take some time.

bishoph, 2017-11-20:
Fixed in latest testing branch with version => 1.4.1

bishoph, 2017-11-21:
Ups, forgot to close the issue :)

closed 2017-11-21 by bishoph

hellowtisch, 2017-11-21:
Hi Martin,

 

the good thing: running python sopare.py -t „test“ does not crash anymore. 

The bad thing: it still throws a lot of errors. Do I need to separately install a jack server that does not bundle with raspbian stretch?

 

pi@raspberrypi:~/sopare $ python sopare.py -t "test"

sopare 1.4.1

ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.front.0:CARD=0'

ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory

ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side

ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'

ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory

ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21

ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'

ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory

ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21

ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround40.0:CARD=0'

ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory

ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround40

ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'

ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory

ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround41

ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'

ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory

ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround50

ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'

[... 50 lines cut ...]
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'

ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory

ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa

ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'

ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory

ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa

Cannot connect to server socket err = No such file or directory

Cannot connect to server request channel

jack server is not running or cannot be started

JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock

JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock

bishoph, 2017-11-21:
You need to clean your ALSA config:
/usr/share/alsa/alsa.conf

Just comment out the stuff which throws errors like this:
#pcm.phoneline cards.pcm.phoneline

Please note that this messages are not SOPARE related :)

Also, please avoid copy'n'paste your personal information into the issues! I've to clean them regularly and manually and you don't want your personal emails to be public I guess, right?!

hellowtisch, 2017-11-21:
I am now on version 1.4.3: The training is still not working, due to some configuration problem with alsa? 

I am able to record sound with the USB microphone using arecord -f a.wav -D sysdefault:CARD=1 and play it back. 

 

pi@raspberrypi:~/sopare $ ./sopare.py -t two

sopare 1.4.3

Cannot connect to server socket err = No such file or directory

Cannot connect to server request channel

jack server is not running or cannot be started

JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock

JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock

ERROR:sopare.recorder:IOError: [Errno -9996] Invalid input device (no default output device)

JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock

hellowtisch, 2017-11-21:
Audio devices seem to be fine…

No other sopare process is running, I confirmed this.

 

Here are the ouputs:

pi@raspberrypi:~/sopare $ python test/test_audio.py

test_audio init...

ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.front.0:CARD=0'

ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory

ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side

ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'

ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory

ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21

ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'

ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory

ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21

ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround40.0:CARD=0'

ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory

ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround40

ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'

ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory

ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround41

ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'

ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory

ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory

ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround50

ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'

ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory

[... 178 lines cut ...]
Error: [Errno -9981] Input overflowed

Excellent. Got all 10240 chunks.

Error: [Errno -9981] Input overflowed

Excellent. Got all 40960 chunks.

Error: [Errno -9981] Input overflowed

 

 

 

Your sopare/config.py recommendations:

 

SAMPLE_RATE = 48000

CHUNK = 1024

THRESHOLD = 2800

bishoph, 2017-11-21:
How many systems do you use? I'm asking because a few comments ago your ALSA output was nice and clean but you got the info that there is no usable device. A few minutes later you are posting some output where ALSA is complaining all over the place but found an usable device...

All I can say is that if ALSA complains and gives you an error "-9996" then there is no device to listen. Simple as that. Maybe you are running another program what listens to the microphone, like Alexa, Google Assistent or whateverisinstalledandrunningonyoursystem.

Personally I would install "lsof" and start looking into the dark holes of the system to figure out what command/program is using the mic:

sudo lsof /dev/snd/*

Hope that helps and gives you a direction, because there is not much more I can do :)