bishoph/sopare #8: Network audio stream

larsomen, 2017-11-22:
This is not An issue, it's Just An question.
Would sopare work with An audio stream? I'm working on a project where the only microphone that is available is on an ip camera.
Thnx in advance!

bishoph, 2017-11-22:
Hi. Not out of the box, but in theory a kind of interface and the necessary adapters would do the job. There are currently two "adapters" but no interface:
1) mic input stream
2) file based stream
I think about it and (potentially) add it to my list as it as the code and the overall architecture would benefit apart from your request. However, at the end it is necessary that you write an individual adapter for your kind of input stream - just to be crystal clear.

jjmeseguerrr, 2018-01-22:
Hi @larsomen,
I want to do just the same as you. If you don't need audio output on the machine where SOPARE is you can use output monitor as audio input using PulseAudio. I've tested this on Ubuntu 14.04, not on a Raspberry Pi.
Play the stream:
`mplayer -novideo rtsp://192.168.0.37:554/unicast`
List sound sources. Look for one ending in ".monitor":
`pactl list short sources`
Set it as default input. In my case it was "auto_null.monitor":
`pactl set-default-source auto_null.monitor`
Regards

bishoph, 2018-02-07:
@larsomen if the solution from @jjmeseguerrr works for you please close the issue. Thanks.

makarov-roman, 2018-08-18:
@bishoph Stream API is still very versatile