bishoph/sopare #9: Python 2.7.12 on Ubuntu gives error

blkmaarx, 2018-01-12:
It seems something goes wrong with the buffering, after the command 

~: python sopare.py -t main

I use Python 2.7.12 on Ubuntu (16.04)

Tracelog below

Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "../sopare/sopare/buffering.py", line 44, in run
    self.proc.check_silence(buf)
  File "../sopare/sopare/processing.py", line 83, in check_silence
    self.stop("stop append mode because time is up")
  File "../sopare/sopare/processing.py", line 52, in stop
    self.prepare.stop()
  File "../sopare/sopare/prepare.py", line 72, in stop
    self.filter_reset()
  File "../sopare/sopare/prepare.py", line 89, in filter_reset
    self.filter.reset()
  File "../sopare/sopare/filter.py", line 48, in reset
    self.queue.put({ 'action': 'reset' })
  File "/usr/lib/python2.7/multiprocessing/queues.py", line 100, in put
    assert not self._closed
AssertionError
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/queues.py", line 268, in _feed
    send(obj)
IOError: [Errno 32] Broken pipe

bishoph, 2018-01-12:
Hi,

what SOPARE version are you using? I'm not 100% sure but this issue should be fixed already in the testing branch since quite some time...

blkmaarx, 2018-01-12:
No, I've tried this in both branches (master 1.4.0 and testing 1.4.0- and both show up as sopare 1.4.0, in the first printed line).

bishoph, 2018-01-12:
Master branch is indeed version 1.4, but testing branch is currently version 1.4.9: 
https://github.com/bishoph/sopare/blob/testing/sopare/version.py

Another indicator that your version does not match the current version is that the line where the error occurs is a blank line in the latest version ;)

Just a guess, but maybe you made some local changes and you need to do a merge (e.g. git pull)

blkmaarx, 2018-01-13:
I didn't correctly checkout testing branch. All perfect now, it is working!

closed 2018-01-13 by blkmaarx