bishoph/sopare #11: Sopare 1.4.9 (testing branch) - Crash of sopare when training voice command

Ringo96, 2018-01-15:
Hello,

so i've started training now, but I run into a new problem.
I've tried to train a new word using ./sopare.py -t "Dere"
But right after i speak the word, sopare crashes with the following error message:

> sopare 1.4.9
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/queues.py", line 268, in _feed
    send(obj)
IOError: [Errno 32] Broken pipe

Maybe you can also help me with that as I would really like to get it running.
Thanks!

Ringo96, 2018-01-15:
It seems like the issue #9 but I'm using the latest version of the testing branch. So all files should be up to date normally...

bishoph, 2018-01-16:
Hmm, I'll try to explain the error:
SOPARE is using multiple CPU cores and sends data through queues. This means that several SOPARE instances are running on a different CPU cores. The queue is like a bridge between the SOPARE instances. The error messages says that one bridge is already taken away and the sending SOPARE instance wanted to use the bridge to send data over. Without a bridge data sending is not possible and results in an error. 

Please do the following:
1) Run the audio test and post the results (python test/test_audio.py)

2) Post your currently used config settings. The [stream] section is important and can normally be used to avoid the error 

3) Check your input before you train and make sure SOPARE is processing what you really want, e.g.:
./sopare.py -w samples/test.raw -~
After the recording play the raw file and play the sound tokens which SOPARE creates in the token/ directory...

Ringo96, 2018-01-16:
[default.ini.txt](https://github.com/bishoph/sopare/files/1636283/default.ini.txt)

Hi,

sooo
**to point 1:**
I tried to start up python test/test_audio.py but it resulted in the error

> Traceback (most recent call last):
  File "test/test_audio.py", line 31, in <module>
    import sopare.audio_factory
ImportError: No module named audio_factory

so i dug a bit in your code and it seems that the issue is the sys.path.append("../sopare").
As far as I could understand from some [threads](https://pythonadventures.wordpress.com/2012/03/31/import-a-file-from-another-directory-of-the-project/) [online](https://stackoverflow.com/questions/714063/importing-modules-from-parent-folder) is that the method you use is fine as long as you start the test_audio.py within the test directory. As soon as you use test/test_audio.py the path points somewhere else and the interpreter cannot resolve the file path anymore.
I'm not sure if this just works this way on my system or if it's a general problem.
So I changed this:

> sys.path.append('../sopare')
import sopare.log
import sopare.config
import sopare.audio_factory

in the test_audio.py to this:

> import os
sys.path.insert(1, os.path.join(sys.path[0], '../sopare'))
import log
import config
import audio_factory

and deleted the _sopare._ startings for all occurences (for ex. sopare.log.log(..) to log.log(..))

This got the script running and the output is following:

> test_audio init...
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
testing different SAMPLE_RATEs ... this may take a while!

> DEBUG:audio_factory:####
Default input device info #####
DEBUG:audio_factory:defaultSampleRate: 44100.0
DEBUG:audio_factory:defaultLowOutputLatency: 0.0087074829932
DEBUG:audio_factory:defaultLowInputLatency: 0.00868480725624
DEBUG:audio_factory:maxInputChannels: 1
DEBUG:audio_factory:structVersion: 2
DEBUG:audio_factory:hostApi: 0
DEBUG:audio_factory:index: 2
DEBUG:audio_factory:defaultHighOutputLatency: 0.0348299319728
DEBUG:audio_factory:maxOutputChannels: 2
DEBUG:audio_factory:name: USB Audio Device: - (hw:1,0)
DEBUG:audio_factory:defaultHighInputLatency: 0.0348299319728
Expression 'paInvalidSampleRate' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2048
Expression 'PaAlsaStreamComponent_InitialConfigure( &self->capture, inParams, self->primeBuffers, hwParamsCapture, &realSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2719
Expression 'PaAlsaStream_Configure( stream, inputParameters, outputParameters, sampleRate, framesPerBuffer, &inputLatency, &outputLatency, &hostBufferSizeMode )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2843
ERROR:audio_factory:Error: [Errno -9997] Invalid sample rate
Expression 'paInvalidSampleRate' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2048
Expression 'PaAlsaStreamComponent_InitialConfigure( &self->capture, inParams, self->primeBuffers, hwParamsCapture, &realSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2719
[... 33 lines cut ...]
CHUNK = 512
THRESHOLD = 100

**2. )** I've attached the default.ini as it would use to much place here, but the [stream] part ist this:

> [stream]
**Read chunk size**
CHUNK = 512
**Sample rate**
SAMPLE_RATE = 48000
**Volume threshold when audio processing starts / silence**
THRESHOLD = 1000
**Silence time in seconds when analysis is called**
MAX_SILENCE_AFTER_START = 1.4
**Time in seconds after the analysis is forced**
MAX_TIME = 2.4
**Start the analysis after reaching LONG_SILENCE**
LONG_SILENCE = 20
**Characteristic length**
CHUNKS = 3072

**3.)** ok thanks for the tip, i'll try as soon as i get the training running.

So can you tell me what I could try to change, to get the training running?
Thanks for your help so far!

bishoph, 2018-01-16:
Great. In terms of the import issue I've no idea why this doesn't work on your system but this is not our main concern right now. Just a quick guess is that you are starting the command from a different directory than I would normally do or the Python version is very different ;) More important is the output and more specific the THRESHOLD. Please change the THRESHOLD in your config to a lower value like shown below. In addition, increase the LONG_SILENCE to something above 30. Delete your current test files (See the quick start, useful commands and usage examples for the commands) and start the training again:

[stream]
CHUNK = 512
SAMPLE_RATE = 48000
THRESHOLD = 300
MAX_SILENCE_AFTER_START = 1.4
MAX_TIME = 2.4
LONG_SILENCE = 50
CHUNKS = 2048

Not only the error should be gone, you should also check what SOPARE is analyzing in the training by following the step 3 from my earlier comment.  I must admit that I'm using SOPARE mostly on several Raspberry Pi's with Raspbian and the time I've tested on a normal PC with Debian was quite a few month ago ;)

Ringo96, 2018-01-16:
I've tried your settings, and you were right, the error is gone now and i was able to train a word. 
Did you remove the "start endless recording" and "stopped recording" lines by the way? Because it doesn't show up when starting sopare, which confused me a bit and caused me thinking that it isn't working.
But it recognises the trained word, so it seems to work :)
I'm still getting alot false positives (for a very short word) and a lot of not recognised times for a rather long one , but it looks quite good, for the fact that I just started training with just a few training cycles and not really tweaked settings and a really shitty self-soldered mic 😀 
I will try your tutorial on better precision now, so things should get better^^

Thank you really a lot for the great help!! Your project is awesome.

bishoph, 2018-01-17:
Thanks and you are welcome :)

The "start endless recording" and "stopped recording" lines show up when you change the log level to >= INFO.

Please note that you have to tweak the settings and test around a lot to get good and reliable results for your environment and requirements. I just wanted to make sure the error you got was not an error I'm need to fix in the code immediately ;) 

As I've changed quite some stuff since my last videos which means you eventually find out that you can't rely on the available information 100% and you need to adapt here and there...

Have fun!

bishoph, 2018-02-07:
Closing issue as there are no bugs or questions open as far as I can see.

closed 2018-02-07 by bishoph

nmenoni, 2020-11-05:
I don't know why did you close it, since I'm using the last version (1.5.0) and still hitting the problem.

nmenoni, 2020-11-05:
I just changed this line of code in line 61 in sopare/sopare/filter.py:

```
old ->       # progessive += progessive * pf
new ->       progessive += progessive * self.cfg.getfloatoption('characteristic', 'PEAK_FACTOR')
```

the error has gone but it is not recognizing any pattern.

bishoph, 2020-11-05:
Last version is 1.5.5, see https://github.com/bishoph/sopare/blob/testing/sopare/version.py