bishoph/sopare #12: Unable to train

sp1d3rino, 2018-02-07:
Hi!

I'm trying to train sopare but after I've ran `./sopare -t test` and I pronounced "test" nothing happens...
I installed all dependencies, and I tested mic with arecord and it works fine. 
I'm using usb mic. Is it necessary to configure the default mic device somewhere in sopare?

bishoph, 2018-02-07:
You need to compile the dict after training and before you get results:

1) Training (./sopare.py -t test)
2) Compiling (./sopare.py -c)
3) Running (./sopare.py -l)

This is of course only extrem quick start. You should configure and optimize according to your environment. Watch the tutorials and more information here: https://www.bishoph.org

Also, check the "Quick start, useful commands and usage examples" in the readme.md :)

closed 2018-02-07 by bishoph

sp1d3rino, 2018-02-07:
Thank you Bishoph!
I will try one more time as soon as possibile