bishoph/sopare #13: it doesn't stop on training

sp1d3rino, 2018-02-07:
hi Martin,
I'm using 1.5.0 
I tried to do sopare.py -t test but it stucks on this log message:
INFO:sopare.recorder:start endless recording

and after I've spoken nothing happens..
I need to do ctrl-c stop stop the script, so I don't have nothing to compile....
any suggestion?

bishoph, 2018-02-07:
First issue: Your THRESHOLD  seems to be too high. The audio_test should give you a THRESHOLD for your environment. Speaking about the test ... can't say anything as some output is missing. The test runs quite some time so you should be patient. If there is a problem you should get an error message. 

Can't see any errors or problems...please close the issue or provide more info.

sp1d3rino, 2018-02-07:
hmm I set THRESHOLD to 100 but I can't train... the issue is that when I start trainig
sopare.py -t test
the script starts, I can ear a low noise into my headphones (so I understand the recording has started)
but the script stops in the following way until I kill it with ctrl-c
INFO:sopare.worker:worker queue runner started
INFO:sopare.buffering:buffering queue runner
INFO:sopare.recorder:start endless recording

^CProcess worker for filtered data:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/home/pi/sopare/sopare/worker.py", line 88, in run
    obj = self.queue.get()
  File "/usr/lib/python2.7/multiprocessing/queues.py", line 117, in get
    res = self._recv()
KeyboardInterrupt
Traceback (most recent call last):
  File "./sopare.py", line 187, in <module>
    main(sys.argv[1:])
  File "./sopare.py", line 108, in main
    recorder.recorder(cfg)
  File "/home/pi/sopare/sopare/recorder.py", line 44, in __init__
    self.recording()
  File "/home/pi/sopare/sopare/recorder.py", line 88, in recording
    buf = self.stream.read(self.cfg.getintoption('stream', 'CHUNK'))
  File "build/bdist.linux-armv7l/egg/pyaudio.py", line 608, in read
    return pa.read_stream(self._stream, num_frames, exception_on_overflow)
KeyboardInterrupt
Process buffering queue:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/home/pi/sopare/sopare/buffering.py", line 40, in run
    buf = self.queue.get()
  File "/usr/lib/python2.7/multiprocessing/queues.py", line 117, in get
    res = self._recv()
KeyboardInterrupt

bishoph, 2018-02-07:
Start SOPARE in debug mode and post whats going on:

./sopare.py -v -t test

bishoph, 2018-02-07:
Ups, hit the wrong button :)

sp1d3rino, 2018-02-07:
sopare 1.5.0
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.front.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
[... cut ...]
DEBUG:sopare.audio_factory:structVersion: 2
DEBUG:sopare.audio_factory:hostApi: 0
DEBUG:sopare.audio_factory:index: 5
DEBUG:sopare.audio_factory:defaultHighOutputLatency: 0.0348299319728
DEBUG:sopare.audio_factory:maxOutputChannels: 32
DEBUG:sopare.audio_factory:name: default
DEBUG:sopare.audio_factory:defaultHighInputLatency: 0.0348299319728
INFO:sopare.buffering:buffering queue runner
DEBUG:sopare.recorder:SAMPLE_RATE: 48000
DEBUG:sopare.recorder:CHUNK: 512
INFO:sopare.recorder:start endless recording

bishoph, 2018-02-07:
There is no audio input SOPARE can use ... so you need to fix your audio settings. Perhaps the input level it too low or if you are using several mics pyaudio is using the wrong one...

sp1d3rino, 2018-02-07:
this is the same thing I'm thinking too 
I just ran arecord -l
card 1: U0x46d0x81b [USB Device 0x46d:0x81b], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

but sopare seems to use CARD=0 .... ?!?!

bishoph, 2018-02-07:
SOPARE uses pyaudio and pyaudio depends on the ALSA configuration. Seems that the default card is "0". I suggest that you fix your alsa.conf as you get quite a lot warnings and info and in addition you can change the default card number...

sp1d3rino, 2018-02-09:
I'm using a usb webcam so I've mic on CARD 1.
Using arecord with "-D plughw:1,0" parameter I can record correctly. Infact if I simply use 
`arecord test.wav` I can't record anything...
I need to set default record CARD 1 so that pyaudio can use it but I don't know how. I tried to modify /usr/share/alsa/alsa.conf and restart alsa with `alsactl kill rescan` without success.

bishoph, 2018-02-09:
Normally you change the card defaults and everything should be fine:

defaults.ctl.card 1
defaults.pcm.card 1

But of course it can be more complicated. Maybe this link helps you: https://superuser.com/questions/626606/how-to-make-alsa-pick-a-preferred-sound-device-automatically

As this issue is not directly SOPARE related and depends on your environment I can't really help.

fabras78, 2018-02-11:
Finally I solved!!! now it works and actually it works like a charm!

so edited /etc/asound.conf in this way 

```
pcm.softvol {
    type softvol
    slave.pcm dmix
    control {
        name Master
        card 0
    }
}

pcm.micboost {
    type route
    slave.pcm dsnoop
    ttable {
        0.0 30.0
        1.1 30.0
    }
}

pcm.!default {
    type asym
    playback.pcm "plug:softvol"
    capture.pcm "plug:micboost"
}

ctl.!default {
    type hw
    card 0
}
```

and after I've ran sopare.py -u that it told me to set to 3200 the threshold.
after that I was able to record, compile and listen and recognize my voice. 

next step use sopare as wake up for my python app that use google speech recognition

thank you, bye!

bishoph, 2018-02-12:
Great, even if the threshold seems to be too high. SOPARE starts the analysis whenever the threshold is reached. So your environment must be very noisy. Anyway, closing the issue.

Have fun!

closed 2018-02-12 by bishoph