bishoph/sopare #14: global name 'pf' is not defined

jvanhaky, 2018-02-21:
Hi, I'm getting the below error message when running ./sopare.py -l -v. Randomly it seems to work for some time but will eventually crash. Here is the output. Any clue what is going wrong?

`pi@raspberrypi:~/sopare $ ./sopare.py -l -v
sopare 1.5.0
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
INFO:sopare.analyze:checking for plugins...
DEBUG:sopare.analyze:loading and initialzing plugins/print
DEBUG:sopare.audio_factory:#### Default input device info #####
DEBUG:sopare.audio_factory:defaultSampleRate: 48000.0
DEBUG:sopare.audio_factory:defaultLowOutputLatency: -1.0
DEBUG:sopare.audio_factory:defaultLowInputLatency: 0.0213333333333
DEBUG:sopare.audio_factory:maxInputChannels: 128
INFO:sopare.worker:worker queue runner started
DEBUG:sopare.audio_factory:structVersion: 2
DEBUG:sopare.audio_factory:hostApi: 0
DEBUG:sopare.audio_factory:index: 5
DEBUG:sopare.audio_factory:defaultHighOutputLatency: -1.0
DEBUG:sopare.audio_factory:maxOutputChannels: 0
DEBUG:sopare.audio_factory:name: default
DEBUG:sopare.audio_factory:defaultHighInputLatency: 0.0213333333333
INFO:sopare.buffering:buffering queue runner
DEBUG:sopare.recorder:SAMPLE_RATE: 44100
DEBUG:sopare.recorder:CHUNK: 512
INFO:sopare.recorder:start endless recording
INFO:sopare.processing:starting append mode
DEBUG:sopare.filter:New window!
Process buffering queue:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/home/pi/sopare/sopare/buffering.py", line 43, in run
    self.proc.check_silence(buf)
  File "/home/pi/sopare/sopare/processing.py", line 74, in check_silence
    self.prepare.prepare(buf, volume)
  File "/home/pi/sopare/sopare/prepare.py", line 125, in prepare
    self.tokenize(meta)
  File "/home/pi/sopare/sopare/prepare.py", line 50, in tokenize
    self.filter.filter(self.buffer, meta)
  File "/home/pi/sopare/sopare/filter.py", line 126, in filter
    chunked_norm = self.get_chunked_norm(nfft)
  File "/home/pi/sopare/sopare/filter.py", line 61, in get_chunked_norm
    progessive += progessive * pf
NameError: global name 'pf' is not defined
INFO:sopare.recorder:Buffering not alive, stop recording
INFO:sopare.recorder:stop endless recording`

bishoph, 2018-02-22:
Thanks, you found an ugly bug. You can avoid the issue by not using the option "PROGRESSIVE_FACTOR" or with the following patch:
<pre>
diff --git a/sopare/filter.py b/sopare/filter.py
index dcacc2e..4f2959a 100644
--- a/sopare/filter.py
+++ b/sopare/filter.py
@@ -58,7 +58,7 @@ class filtering():
         i = self.cfg.getintoption('characteristic', 'MIN_PROGRESSIVE_STEP')
         for x in range(0, nfft.size, i):
             if (self.cfg.hasoption('characteristic', 'START_PROGRESSIVE_FACTOR') and x >= self.cfg.getfloatoption('characteristic', 'START_PROGRESSIVE_FACTOR')):
-                progessive += progessive * pf
+                progessive += progessive * self.cfg.getfloatoption('characteristic', 'START_PROGRESSIVE_FACTOR')
</pre>

bishoph, 2018-02-22:
One comment in regards to the "PROGRESSIVE_FACTOR" option. This option helps to reduce and compress frequencies above a certain frequency. This means if you have a large range of frequencies lets say from 100 to 24000 and you get only very small blips above 2000 than this option reduces the number of processed frequencies above a certain frequency with the help of progression. Maybe I will move the option to the experimental section as it is not used in any of my environments and really an exotic use case (and it is not covered by test cases yet - but this is entirely my fault ;)

jvanhaky, 2018-02-27:
It probably got a bit better but now another error of buffer overflowing appears after a short period after starting sopare with -l -v.

```
... 0.05487131283140254, 0.04829192763683348, 0.017683463175775137], 'token_peaks': [22949, 17893, 40805, 98891, 22459, 47387, 74353, 36136, 86510, 44949, 34608, 117472]}
DEBUG:sopare.worker:meta = [{'adapting': 117472, 'token_peaks': [22949, 17893, 40805, 98891, 22459, 47387, 74353, 36136, 86510, 44949, 34608, 117472], 'pos': 12, 'volume': 469, 'token': 'token', 'silence': 0}]
WARNING:sopare.recorder:stream read error [Errno -9981] Input overflowed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed
...
```

Here is git diff from my config file (only minor changes to original config file from git):
```
pi@raspberrypi:~/sopare $ git diff
diff --git a/config/default.ini b/config/default.ini
index 9c2c26c..564c033 100644
--- a/config/default.ini
+++ b/config/default.ini
@@ -5,13 +5,13 @@
 [stream]
 
 # Read chunk size
-CHUNK = 512
+CHUNK = 256
 
 # Sample rate
 SAMPLE_RATE = 48000
 
 # Volume threshold when audio processing starts / silence
-THRESHOLD = 380
+THRESHOLD = 220
 
 # Silence time in seconds when analysis is called
 MAX_SILENCE_AFTER_START = 1.4
@@ -38,15 +38,15 @@ CHUNKS = 3072
 # Progressive value is used if you want to pack not
 # so relevant frequencies
 PROGRESSIVE_FACTOR = 0
-START_PROGRESSIVE_FACTOR = 600
+START_PROGRESSIVE_FACTOR = 5000
 MIN_PROGRESSIVE_STEP = 5
 MAX_PROGRESSIVE_STEP = 5
 
 # Specifies freq ranges that are kept for further
 # analysis. Freq outside of the ranges are set to zero.
 # Human language can be found between 20 and 5000.
-LOW_FREQ = 20
-HIGH_FREQ = 600
+LOW_FREQ = 100
+HIGH_FREQ = 5000
 
 # Make use of Hann window function
 HANNING = true
```

bishoph, 2018-02-27:
What OS are you using?
Did you run the audio test (python test/test_audio.py)? If yes, post your recommendations. If not, run the test and change the config accordingly to the recommendations.

jvanhaky, 2018-02-27:
**Device and OS info:**

cat /proc/device-tree/model
Raspberry Pi 2 Model B Rev 1.1

cat /etc/os-release
Raspbian GNU/Linux 9 (stretch).

uname -a
Linux raspberrypi 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux

**test**
I have run the tests and they mostly succeed. However the same overflow errors appears there among successful tests. I have tried various configurations and nothing works perfectly. If I lower the frequency range it works longer. Also if I set too low threshold causing it to analyze "constantly", it will overflow which I think it shouldn't.

I'm using a cheap USB microphone from ebay but it seems to work in other programs.
https://www.ebay.com/itm/New-SF-555B-Black-Computer-Studio-Super-USB-2-0-Microphone-wire-Mic-professional/302510772560?epid=3009232933&hash=item466f0c1150:g:AfAAAOSwMmBV4-60

Here is the output from test:

```
pi@raspberrypi:~/sopare $ python test/test_audio.py 
test_audio init...
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
testing different SAMPLE_RATEs ... this may take a while!

DEBUG:sopare.audio_factory:#### Default input device info #####
DEBUG:sopare.audio_factory:defaultSampleRate: 48000.0
DEBUG:sopare.audio_factory:defaultLowOutputLatency: -1.0
DEBUG:sopare.audio_factory:defaultLowInputLatency: 0.0213333333333
DEBUG:sopare.audio_factory:maxInputChannels: 128
DEBUG:sopare.audio_factory:structVersion: 2
DEBUG:sopare.audio_factory:hostApi: 0
DEBUG:sopare.audio_factory:index: 5
DEBUG:sopare.audio_factory:defaultHighOutputLatency: -1.0
DEBUG:sopare.audio_factory:maxOutputChannels: 0
DEBUG:sopare.audio_factory:name: default
DEBUG:sopare.audio_factory:defaultHighInputLatency: 0.0213333333333
testing different CHUNK sizes ... this may take a while!

Excellent. Got all 5120 chunks.
Excellent. Got all 10240 chunks.
Excellent. Got all 20480 chunks.
Excellent. Got all 40960 chunks.
Excellent. Got all 81920 chunks.
Excellent. Got all 5120 chunks.
Excellent. Got all 10240 chunks.
Excellent. Got all 20480 chunks.
Excellent. Got all 40960 chunks.
Excellent. Got all 81920 chunks.
Excellent. Got all 5120 chunks.
Excellent. Got all 10240 chunks.
Excellent. Got all 20480 chunks.
Excellent. Got all 40960 chunks.
Excellent. Got all 81920 chunks.
Excellent. Got all 5120 chunks.
Excellent. Got all 10240 chunks.
Excellent. Got all 20480 chunks.
Excellent. Got all 40960 chunks.
Excellent. Got all 81920 chunks.
Excellent. Got all 5120 chunks.
Excellent. Got all 10240 chunks.
Excellent. Got all 20480 chunks.
Excellent. Got all 40960 chunks.
Excellent. Got all 81920 chunks.
Excellent. Got all 5120 chunks.
Excellent. Got all 10240 chunks.
Excellent. Got all 20480 chunks.
Excellent. Got all 40960 chunks.
Excellent. Got all 81920 chunks.
Excellent. Got all 5120 chunks.
Excellent. Got all 10240 chunks.
Excellent. Got all 20480 chunks.
Excellent. Got all 40960 chunks.
Excellent. Got all 81920 chunks.
Excellent. Got all 5120 chunks.
Excellent. Got all 10240 chunks.
Excellent. Got all 20480 chunks.
Error: [Errno -9981] Input overflowed
ERROR:sopare.audio_factory:Error: Stream not open
Error: [Errno -9981] Input overflowed
ERROR:sopare.audio_factory:Error: Stream not open

Your sopare/config.py recommendations:

SAMPLE_RATE = 48000
CHUNK = 512
THRESHOLD = 900
ERROR:sopare.audio_factory:Error: Stream not open
```

bishoph, 2018-02-27:
Try all of the tested sample rates (8000, 11025, 12000, 16000, 22050, 32000, 44100) in combination with a CHUNK size of 512 and check if the warning is gone. If not, the message is annoying, but I can't do much as this warning comes from the underlying pyAudio lib.

However, as the original error report seems to be fixed I close this issue.

closed 2018-02-27 by bishoph

aliomar786, 2019-04-15:
the same issue happend to me when i increased the frequency range ? i didnt under the solution you provided so if you could please tell me how to get this matter solved