bishoph/sopare #17: Idea: use smartphone as a microphone

cs301cs301, 2018-04-07:
Since everyone has a smartphone, is there a way that the samrtphone can act as mic for sopare? Can you build such a feature? One advantage would be, that no one has to mess with the ALSA configuration.

bishoph, 2018-04-08:
As SOPARE depends on ALSA and pyAudio this is beyond the scope of this project. In theory you need to pipe/redirect/route the phones mic input to ALSA, make it the default and it should work.

onacvooe, 2019-04-03:
Just my 2 cents about this question. I am not a programmer (yet). But I know from another project which also does voice recognition via python. They have a android app which links to a little webserver and when you press record in the android app it sends the voice command to the webserver and executes the voice command. 

I *think* that it sends an wav/mp3/sound file to the webserver which in turn *speaks* to the voice recognition program. I hope this helps and somebody might get a "AHAH"! moment ;-) Happy coding everyone!