bishoph/sopare #20: OverflowError

cs301cs301, 2018-04-17:
Recording works, in mono, too. But there is an overflow error...

sopare 1.5.2
DEBUG:sopare.audio_factory:#### Default input device info #####
DEBUG:sopare.audio_factory:defaultSampleRate: 44100.0
DEBUG:sopare.audio_factory:defaultLowOutputLatency: 0.0087074829932
DEBUG:sopare.audio_factory:defaultLowInputLatency: 0.00580498866213
DEBUG:sopare.audio_factory:maxInputChannels: 2
DEBUG:sopare.audio_factory:structVersion: 2
DEBUG:sopare.audio_factory:hostApi: 0
DEBUG:sopare.audio_factory:index: 2
DEBUG:sopare.audio_factory:defaultHighOutputLatency: 0.0348299319728
INFO:sopare.worker:worker queue runner started
DEBUG:sopare.audio_factory:maxOutputChannels: 2
DEBUG:sopare.audio_factory:name: seeed-2mic-voicecard: – (hw:1,0)
DEBUG:sopare.audio_factory:defaultHighInputLatency: 0.0348299319728
INFO:sopare.buffering:buffering queue runner
DEBUG:sopare.recorder:SAMPLE_RATE: 16000
DEBUG:sopare.recorder:CHUNK: 512
INFO:sopare.recorder:start endless recording
INFO:sopare.processing:starting append mode
DEBUG:sopare.filter:New window!
Process buffering queue:
Traceback (most recent call last):
File „/usr/lib/python2.7/multiprocessing/process.py“, line 258, in _bootstrap
self.run()
File „/home/sopare/sopare/buffering.py“, line 43, in run
self.proc.check_silence(buf)
File „/home/sopare/sopare/processing.py“, line 74, in check_silence
self.prepare.prepare(buf, volume)
File „/home/sopare/sopare/prepare.py“, line 126, in prepare
self.tokenize(meta)
File „/home/sopare/sopare/prepare.py“, line 50, in tokenize
self.filter.filter(self.buffer, meta)
File „/home/sopare/sopare/filter.py“, line 126, in filter
chunked_norm = self.get_chunked_norm(nfft)
File „/home/sopare/sopare/filter.py“, line 62, in get_chunked_norm
i += int(progessive)
OverflowError: cannot convert float infinity to integer
INFO:sopare.recorder:Buffering not alive, stop recording
INFO:sopare.recorder:stop endless recording

same error if i read a .wav or .raw file.

cs301cs301, 2018-04-17:
here is output of ./sopare.py -l (after creating the dic!)

sopare 1.5.2
DEBUG:sopare.audio_factory:#### Default input device info #####
DEBUG:sopare.audio_factory:defaultSampleRate: 44100.0
DEBUG:sopare.audio_factory:defaultLowOutputLatency: 0.0087074829932
DEBUG:sopare.audio_factory:defaultLowInputLatency: 0.00580498866213
DEBUG:sopare.audio_factory:maxInputChannels: 2
DEBUG:sopare.audio_factory:structVersion: 2
INFO:sopare.worker:worker queue runner started
DEBUG:sopare.audio_factory:hostApi: 0
DEBUG:sopare.audio_factory:index: 2
DEBUG:sopare.audio_factory:defaultHighOutputLatency: 0.0348299319728
DEBUG:sopare.audio_factory:maxOutputChannels: 2
DEBUG:sopare.audio_factory:name: seeed-2mic-voicecard: - (hw:1,0)
DEBUG:sopare.audio_factory:defaultHighInputLatency: 0.0348299319728
INFO:sopare.buffering:buffering queue runner
DEBUG:sopare.recorder:SAMPLE_RATE: 16000
DEBUG:sopare.recorder:CHUNK: 512
INFO:sopare.recorder:start endless recording
INFO:sopare.processing:starting append mode
DEBUG:sopare.filter:New window!
DEBUG:sopare.worker:characteristic = 0 {'volume': 15244, 'peaks': [0, 5, 10, 14, 15, 65], 'df': 94, 'dfm': 3450866, 'fc': 606.8, 'zcr': 0, 'norm': [0.10935133984390594, 0.09668085496247761, 0.08568011987889064, 0.07981646849926544, 0.07579406698428724, 0.10141618831859614, 0.09112914012921405, 0.08574975463039335, 0.08355476306683064, 0.09170469326066764, 0.1024780876031649, 0.08853449417745059, 0.08909318594521487, 0.09425561899235321, 0.10996939732324511, 0.10584376309694611, 0.09209846996422301, 0.08113150133391538, 0.09695112424869685, 0.09530088457052928, 0.08448007831820287, 0.07304737030692092, 0.07148212652262659, 0.08630900894271505, 0.09186933768208519, 0.08062855740785001, 0.07309460656378408, 0.06881377925096513, 0.08232832261613177, 0.07371180649332493, 0.0658948679173174, 0.06089447312973458, 0.06737106731595582, 0.07429351700211172, 0.06406816912447658, 0.06389632792656252, 0.06515967021298304, 0.07524940010212779, 0.07339427082918283, 0.0706658022702824, 0.06295215011425874, 0.0669735516134708, 0.0792571233855015, 0.07169310492131836, 0.06800919796330102, 0.06463644536588845, 0.07525586851475194, 0.07795921172471956, 0.07797170031863038, 0.07187257945068373, 0.07458248003369185, 0.08738685360881107, 0.0775099921396784, 0.07768061037512941, 0.07588614443694289, 0.07867628711549098, 0.0833174791891067, 0.07441153539548515, 0.0754122778614868, 0.0793913230393574, 0.09133886205248834, 0.09256928373149591, 0.09065872006048002, 0.08778888946284344, 0.09329678951458543, 0.10415857824126121, 0.09526120962243795, 0.09085291243232842, 0.08629824673571677, 0.08952480256528116, 0.08454422151724605, 0.07296522963994471, 0.07378775277985157, 0.07416038928890803, 0.08366377737463214, 0.0760469056321619, 0.0731344455630373, 0.06393814750379602, 0.07273478375299412, 0.07371059888133571, 0.059535977415674526, 0.05916243328459496, 0.06107933266692879, 0.06469666218605913, 0.0688150453814011, 0.05905253122327085, 0.05029775041368453, 0.052112837057670496, 0.0450717709595119, 0.05387515519917973, 0.05008519473491902, 0.062091625064053094, 0.06726175528166688, 0.07367858987141565, 0.07205888749244181, 0.0640543785587704, 0.06916305755700254, 0.07865179031698487, 0.07476968899677269, 0.08075906792574464, 0.06967338900696268, 0.0796601448643674, 0.08643364420911445, 0.08509853566935595, 0.07501182623394967, 0.07296493459030325, 0.08021708921344327, 0.07212012377117118, 0.07489834416589601, 0.06564933599392368, 0.06929721722088146, 0.07233945518262634, 0.0627071244564514, 0.06691719735376953, 0.06393447872421285, 0.06958195410831546, 0.07180098435154171, 0.06208468955369211, 0.06756139629623625, 0.0685971494408684, 0.07285504220440277, 0.06490563845424517, 0.06729469635720292, 0.06385118066951623, 0.07305733211613397, 0.07544207283356448, 0.0612606240827792, 0.06784278161814858, 0.06059310315782893, 0.06950470543623172, 0.072004989525873, 0.06487444588059818, 0.06611020115130803, 0.06488421460948156, 0.06681770942932894, 0.059262885436620935, 0.06100664897111068, 0.05585155062408036, 0.0591857312274422, 0.05703775770733047, 0.05601785777351152, 0.05832108080154758, 0.056193189110274384, 0.061964883214534625, 0.05585671487243162, 0.062052956441008285, 0.06691403234952896, 0.0648887439973834, 0.07128853883419116, 0.06904319016964103, 0.05499659805258823, 0.05734047849989595, 0.06040199414237221, 0.05544659397068725, 0.05933247426051445, 0.06118421662658103, 0.06217903433052576, 0.06036954511798793, 0.05978885937791951, 0.060577011399243444, 0.055652580835671074, 0.05742427154231599, 0.059544674028445536, 0.05810455283550348, 0.055243416733493166, 0.052497772184369226, 0.059628449675705256, 0.06298490750663643, 0.060073122418440046, 0.0534539995455288, 0.0517934130240308, 0.05191583950953188, 0.05091296276443595, 0.052561474388443406, 0.049081531529754925, 0.05251169885381907, 0.061469249010661295, 0.0528700314418125, 0.043185529624302256, 0.05389352199533596, 0.0652851608334452, 0.06482267765824003, 0.06790610547161154, 0.06399140457852302, 0.063809932504493, 0.05956063178352729, 0.05238086276437654, 0.048102478446353875, 0.04869939632126553, 0.05730459616451783, 0.06311737744998434, 0.061999637119864205, 0.061592305846516426, 0.05855716473278354, 0.05437198789585113, 0.05038954536198432], 'token_peaks': [80195, 226052, 396619, 3275099, 6303196, 6252473]}
[... cut ...]
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/multiprocessing/util.py", line 325, in _exit_function
    p.join()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 145, in join
    res = self._popen.wait(timeout)
  File "/usr/lib/python2.7/multiprocessing/forking.py", line 154, in wait
    return self.poll(0)
  File "/usr/lib/python2.7/multiprocessing/forking.py", line 135, in poll
    pid, sts = os.waitpid(self.pid, flag)
KeyboardInterrupt

bishoph, 2018-04-17:
Ok, thx. The issue is fixed so I close the issue for now.

You need to make sure to add the option 

SIMILARITY_ZERO_CROSSING_RATE 

to the "Calculation basis" options and to your config file

in order to avoid the above "error" message. The option was introduced with one of the last commits.

closed 2018-04-17 by bishoph