bishoph/sopare #24: Question: Do You Place All Your Code in The Output Folder?

snwdrft, 2018-05-16:
From watching your video on the magic mirror, it seemed like you compiled all your code into a folder called events in the output folder (/plugins/print/events/_init_.py). Is that the way to do it, or could you use an external compiler or IDE to do the same job?

bishoph, 2018-05-17:
All folders in the plugin folder of SOPARE are python modules or packages. Here are links with helpful information:

https://www.bishoph.org/sopare-architecture-and-plugins/
https://docs.python.org/2/tutorial/modules.html

Of course you can use a text editor or an IDE to create and edit those files ;)

closed 2018-05-17 by bishoph

snwdrft, 2018-05-21:
I've tried to see if the run function would print the values that are being produced by the program, but nothing seems to show up. It works perfectly fine in the terminal command line, but nothing shows up in the python shell.