bishoph/sopare #25: How Do You Write Readable_Results To a File?

snwdrft, 2018-06-03:
I'm trying to get the print plugin to write to a file, but it doesn't seem to work. 

```
file = "./output.txt"

def run(readable_results, data, rawbuf):
    output = open(file, "a")
    if(len(readable_results) < 1 ):  
        output.write('no response')
    elif('weather' in readable_results):
        output.write('weather')
    elif('news' in readable_results):
        output.write('news')
    elif('video' in readable_results):
        output.write('video')
    output.write("\n")
    output.close()     

```

bishoph, 2018-06-04:
Without further information I have really no idea how to help :) You should provide information about the issue, error messages or whatever helps.  See
https://leantesting.com/write-good-bug-report/
for more details ;)

snwdrft, 2018-06-04:
Alright sorry for the lack of information, I'm trying to change a content of a web app based on voice commands, The only problem is connecting the two plugins, A simple solution was to pipe the values from readable_results from sopare to a text file and have the other plugin read it by threading intervals. I've tried a subprocess command but ended up with a "/bin/sh 1: permission denied" from accessing __init__.py. I tried putting that code up in __init__.py but it doesn't work and also tried putting the entire code inside __init__.py but that overrides sopare.py. 

for context: https://github.com/dddomodossola/remi

bishoph, 2018-06-05:
A bash "permission denied" is nothing where I can help in terms of a fix or a code change. My guess is that the process that throws the error does not has the appropriate permissions. You should use a messaging server to solve this (e.g. https://github.com/mqtt/mqtt.github.io/wiki) or maybe use a directory/file like "/tmp/" where all corresponding processes are able to read/write.

closed 2018-06-05 by bishoph