bishoph/sopare #28: application hangs in "loop forever" mode at high load scenarios

erics465, 2018-06-11:
Hi there,
after setting up the environment following the instructions on your blog (https://www.bishoph.org/step-by-step-raspberry-pi-offline-voice-recognition-with-sopare/), i really like the project. However, I've found a major problem that blocks further use.

When letting `./sopare.py -l` run for a longer time, at a certain point the script hangs. This is noticeable due to is high CPU usage (a full core) and the fact that no more audio is processed (no output in the log). It seems that things like music which are above the THRESHOLD level and therefore trigger a continuous activity of the application cause this problem after a certain amount of time. However, this does not occur for short periods of sound above this level.

I'm running a freshly pulled version of sopare on a raspberry pi 2 with a USB sound card that has a microphone attached to it. In normal situations, the detection works flawlessly.

The following stacktrace shows up when interrupting the application with CTRL+C:
```
^CTraceback (most recent call last):
  File "./sopare.py", line 187, in <module>
    main(sys.argv[1:])
  File "./sopare.py", line 108, in main
    recorder.recorder(cfg)
  File "/home/pi/dev/sopare/sopare/recorder.py", line 44, in __init__
    self.recording()
  File "/home/pi/dev/sopare/sopare/recorder.py", line 95, in recording
    self.logger.warning("stream read error "+str(e))
KeyboardInterrupt
Process buffering queue:
Process worker for filtered data:
Traceback (most recent call last):
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
    self.run()
  File "/home/pi/dev/sopare/sopare/worker.py", line 88, in run
  File "/home/pi/dev/sopare/sopare/buffering.py", line 40, in run
    buf = self.queue.get()
  File "/usr/lib/python2.7/multiprocessing/queues.py", line 117, in get
    obj = self.queue.get()
  File "/usr/lib/python2.7/multiprocessing/queues.py", line 117, in get
    res = self._recv()
    res = self._recv()
KeyboardInterrupt
KeyboardInterrupt
```

bishoph, 2018-06-11:
This is indeed possible if lots of trained sounds exists with the same start indicator and with high precision options. But the system always goes into normal mode after some time. Please note that you can avoid this by reducing the amount of trained sounds and normally also with some config options. Please note that the above output does not show any error and in addition you did not provide any steps to reproduce nor further information...so I'm currently looking into my crystal ball and hope that a  planned optimization helps and you can close the issue ;)

erics465, 2018-06-11:
Thank you for your answer.
I will test if the system will at some point work like it should again. But I only have one pattern with less than ten training rounds. I don't think that this is that much...
I'm sorry that I didn't provide any further steps, but I've read some comments in your blog stating the same issue. That's why I thought that this issue is already known.

bishoph, 2018-06-12:
I don't see this issue as a big problem but I've already some idea for an optimization. I have several systems running 24/7 just fine with lots of background noise such as loud music. 100% CPU is normal for a multi core system like the Raspi 2/3. You can run "top -d 1" and press "1" to see all cores working. One of my test cases is a stress test and the recovery. Of course you can force this issue by using a very low threshold and very high values to start/force the analysis - but as I've written this can be avoided by adjusting the config ;)

GitJer, 2019-12-16:
I'm having the same problem as @erics465. When sopare is working well and there is no sound the load is only a couple of percent on a RPI4. When I speak the load goes up to say 50% but when I am silent again the load drops to low percentages. However, after some time (sometimes seconds, sometimes a minute) the load suddenly shoots up to 100% and sopare freezes. It doesn't matter if I am silent or speaking. When I stop the script with ctrl-c I often get the same message as erics465.

bishoph, 2019-12-17:
Do you have log files with errors or any steps of how to reproduce?

Other than that, it's possible that SPOPARE processes input from the mic you can't hear. Most likely this issue is a config/environment issue. Try to increase the THRESHOLD (e.g. 1000 or even more) and try again with a newly trained word, before delete dict and raw files. If the load is normal after that and you don't get errors try to figure out what filter/config options fit best for your environment...

GitJer, 2019-12-17:
There are only few projects where the owner responds so quickly! Thanks!

I have switched on logging and at the moment the load goes up to 100% I noticed the following warning:

> WARNING:sopare.recorder:stream read error [Errno -9981] Input overflowed
> WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed

The line about the closed Stream repeats until I stop the script.

After some debugging I found that the problem is in recorder.py at line 89: 

> buf = self.stream.read(self.cfg.getintoption('stream', 'CHUNK'))

Which is caused by:

>   File "/usr/lib/python2.7/dist-packages/pyaudio.py", line 608, in read
>     return pa.read_stream(self._stream, num_frames, exception_on_overflow)
> IOError: [Errno -9981] Input overflowed

After looking at the pyaudio.py file I noticed that there is an option to not throw an exception. If I change line 89 into:

> buf = self.stream.read(self.cfg.getintoption('stream', 'CHUNK'), exception_on_overflow=False)

sopare runs fine.

bishoph, 2019-12-17:
Hmm, ignoring the overflow error reduces the load. Sounds a bit odd. Ignoring CHUNKS reduces maybe the accuracy as input gets lost. If this is just one piece it should not affect the process but the load drop indicates something different...

Did you try to choose another SAMPLE_RATE and/or a CHUNK size to avoid the error completely? Asking because this is normally the way to go as this issue shows: https://github.com/bishoph/sopare/issues/32 

Have fun :)

GitJer, 2019-12-17:
I played with CHUNK values from 128 to 8192 but that did not change anything. 

config.py suggested 48000. An hour ago I only changed the SAMPLE_RATE to 44100 but that still caused the problem. I now have tried some other values as well and it does make a difference: I successfully tried 8000 and 22050, but 32000 fails. 

Still, stopping of the overflow error may be handy to have anyway.