bishoph/sopare #32: mic stops scanning

classicryder, 2018-07-26:
i run sopare with -v option and after some time, the mic stops scanning..
right before the mic stops scanning, it shows these messages:
WARNING:sopare.recorder:stream read error [Errno -9981] Input overflowed
WARNING:sopare.recorder:stream read error [Errno -9988] Stream closed

and if it runs without -v option (after some time), the error is:
ERROR:sopare.audio_factory:Error: Stream not open

classicryder, 2018-07-27:
Is this pyaudio problem?

bishoph, 2018-07-28:
Yes, seems like either a pyaudio problem or a wrong sample rate / chunk size. Are you using the suggestions from the audio test? Please post the output from the audio test

<pre>
python test/test_audio.py
</pre>

classicryder, 2018-07-29:
here is my current default.ini and pyaudio test output

#########################################################
# Stream prep and silence configuration options #########
#########################################################

[stream]

# Read chunk size
CHUNK = 512

# Sample rate
SAMPLE_RATE = 48000

# Volume threshold when audio processing starts / silence
THRESHOLD = 300

# Silence time in seconds when analysis is called
MAX_SILENCE_AFTER_START = 1

# Time in seconds after the analysis is forced
MAX_TIME = 2.4

# Start the analysis after reaching LONG_SILENCE
LONG_SILENCE = 20

# Characteristic length
CHUNKS = 3072

#########################################################
# Characteristic configuration options ##################
#########################################################

[characteristic]

# Steps boil down the data into smaller chunks of data.
# Smaller steps mean more precision but require
# normally more learned entries in the dictionary.
# Progressive value is used if you want to pack not
# so relevant frequencies
PROGRESSIVE_FACTOR = 0
START_PROGRESSIVE_FACTOR = 600
MIN_PROGRESSIVE_STEP = 5
MAX_PROGRESSIVE_STEP = 5

# Specifies freq ranges that are kept for further
# analysis. Freq outside of the ranges are set to zero.
# Human language can be found between 20 and 5000.
LOW_FREQ = 20
HIGH_FREQ = 600

# Make use of Hann window function
HANNING = true

# Range factor for peaks
PEAK_FACTOR = 0.7

#########################################################
# Compare configuration options #########################
#########################################################

[compare]

# Min. number of tokens to identify the beginning of a word
MIN_START_TOKENS = 3

# Min. value for potential beginning of a word
MARGINAL_VALUE = 0.7

# Minimal similarity across all comparison to
# identify a complete word across all tokens
MIN_CROSS_SIMILARITY = 0.8

# Calculation basis or token/word comparison
SIMILARITY_NORM = 0.6
SIMILARITY_HEIGHT = 0.4
SIMILARITY_DOMINANT_FREQUENCY = 0

# Number of best matches to consider.
# Value must be > 0
# If not specified or value < 1 value is set to 1
NUMBER_OF_BEST_MATCHES = 2

# Min. distance to keep a word
MIN_LEFT_DISTANCE = 0.9
MIN_RIGHT_DISTANCE = 0.7

# Use given number as results to assembly result
# 0 for all predictions
MAX_WORD_START_RESULTS = 2
MAX_TOP_RESULTS = 3

# Enable or disable strict length check for words
STRICT_LENGTH_CHECK = true
# Value to soften the strict length check a bit to still
# get quite precise results but to be less strict
STRICT_LENGTH_UNDERMINING = 2

# Short term memory retention time in seconds. Zero to disable STM
STM_RETENTION = 1.2

# Fill result percentage
# 0.5 means that half of the values can by empty to still get valid results
# A lower value should theoretically avoid false positives
FILL_RESULT_PERCENTAGE = 0.1

#########################################################
# Misc configuration options ############################
#########################################################

[misc]

# Loglevel (CRITICAL, ERROR, WARNING, INFO, DEBUG)
LOGLEVEL = ERROR

#########################################################
# Experimental configuration options ####################
#########################################################

[experimental]

# Additional FFT analysis and comparison for CHUNKS/2 length
FFT_SHIFT = false

--------------------------------------------------------------------------------------------------------------------

Excellent. Got all 5120 chunks.
Excellent. Got all 10240 chunks.
Excellent. Got all 20480 chunks.
Excellent. Got all 40960 chunks.
Excellent. Got all 81920 chunks.
Excellent. Got all 5120 chunks.
Excellent. Got all 10240 chunks.
Excellent. Got all 20480 chunks.
Excellent. Got all 40960 chunks.
Excellent. Got all 81920 chunks.

Your sopare/config.py recommendations:

SAMPLE_RATE = 48000
CHUNK = 512
THRESHOLD = 300

bishoph, 2018-07-30:
As the test just runs fine I have no idea. the error is thrown by ALSA and means that the system can't keep up reading from the mic. I only give you some ideas: 

1) try lower sample rates from starting with the lowest working sample rate from the test output
2) try a higher chunk size

If this does not help try a different mic. I also have one cheap USB mic which throws this kind of error but in my case the system still keeps working and it only affects the accuracy a little bit.
There are endless error reports for ALSA, Raspberry Pi and around the errors "-9981" and "-9988" maybe some solutions are working for you.

closed 2018-07-30 by bishoph

classicryder, 2018-07-30:
Just curious, what mic are u currently using for the 24/7 light in your home? R u suggesting me to change the mic or the usb interface?

bishoph, 2018-07-30:
See https://www.bishoph.org/sopare-basic-usage-voice-controlling-a-magic-mirror/ for details about hardware.

My suggestion is that you should first try to lower the sample rate, then to use a higher chunk size and if all that does not work to consult the internet about your sound card and the specific errors ... the very last step is to think about different hardware in my humble opinion.

classicryder, 2018-07-31:
After changing the default.ini config and re train the sound pattern, i got this message:

The followinf characteristic is < 2

Is it a problem?

My trained sound pat can't be recognized too

bishoph, 2018-07-31:
Yes, the characteristic is too short for a recognition. Did you ever try to suggested combination

SAMPLE_RATE = 48000
CHUNK = 512

?

classicryder, 2018-07-31:
Yes i use that value and then i got the error from pyaudio, then i tried to use bigger chunk like you suggested then this happened 😄

bishoph, 2018-07-31:
Sorry I was confused with your CHUNKS value of 3072. Seems the heat melts my brain slowly. 

Try to increase the MAX_SILENCE_AFTER_START value to 2 seconds and try again. If this does not help I have no idea why this does not work for you :(

classicryder, 2018-07-31:
Wait, when you said bigger chunk, are you referring to chunk=512 or chunks=3072?

bishoph, 2018-07-31:
Bigger CHUNK :)

classicryder, 2018-07-31:
I noticed that using bigger chunk (1000) would eliminate the error but the trained word is messed up (the characteristic < 2 message, everytime this message appear, the trained word cant be recognized, and increasing the max silence after start parameter doesnt help)

If i use 512 on the chunk, the trained word works but in like 1-2 hours the pyaudio error appears

classicryder, 2018-08-01:
Im using Rpi 3 model B+

bishoph, 2018-08-02:
The error comes from ALSA/pyaudio and I would try to fix the underlying issue and not ignore the error message as I doubt that this will help you long term (even if there is an option to pass to pyaudio to ignore the message though). There are hundreds of threads on StackOverflow around this messages - maybe you find something helpful...

classicryder, 2018-08-02:
I know.. tons of them are also suggesting bypassing this error message.. even if i use other method, i still need to figure out which one in your codes/files that actually start the recording.. im curious what file is it? And where is the main?

Sorry if this is a silly question

bishoph, 2018-08-02:
I hope you know what you are doing: https://github.com/bishoph/sopare/blob/master/sopare/audio_factory.py

:)

classicryder, 2018-08-03:
Thanks sir :)

Apart from that, when i use chunk = 1000, it runs smoothly without stream closed/input overflow error, but i cant train the sound pattern, even the raw file is not showing up, is there any missing steps?

bishoph, 2018-08-03:
Maybe a too short time span. Try to increase MAX_SILENCE_AFTER_START, MAX_TIME  and maybe LONG_SILENCE. 

I would use something like these parameters and adjust after you got a working base:
MAX_SILENCE_AFTER_START= 2
MAX_TIME  = 3
LONG_SILENCE = 100

classicryder, 2018-08-08:
So, it seems like by using these settings, the problem was solved
CHUNK = 1000
CHUNKS = 4000
RATE = 44100

but.. sometimes after a reboot, the trained word cannot be recognized, is there anyway to start sopare.py in terminal (i need to see the verbose output) when the pi gets booted up?
i tried modding the cron job so that it will be started after booting, but it doesnt show the terminal output

Thanks

classicryder, 2018-08-08:
no, what i need is when the pi booted up, it automatically opens a terminal, then run the sopare.py so that i can see whats going on directly from the splash screen :)

bishoph, 2018-08-08:
Then open a terminal and use

<pre>
tail -f error.log
</pre>

:)

Other than that there are plenty of possibilities with pure Unix/Linux commands...