bishoph/sopare #33: divide by zero

cs301cs301, 2018-07-29:
Hi,

i set CHUNKS = 1024 and sometimes this occurs in training (mostly in the third training round of a word)...

/home/sopare/sopare/filter.py:138: RuntimeWarning: divide by zero encountered in log10
  shift_nfft = numpy.log10(nfft)**2
/home/sopare/sopare/filter.py:143: RuntimeWarning: invalid value encountered in divide
  shift_nfft = numpy.tanh(shift_nfft/shift_nam)

any idea?

thx

bishoph, 2018-07-30:
This is a numpy warning message. I have to check if the input is zero before passing it to numpy. Will fix it with one of the next commits. Thanks for reporting :)

bishoph, 2019-03-31:
Should be fixed with commit ca692ff in testing branch (version >= 1.5.4)

closed 2019-03-31 by bishoph