bishoph/sopare #35: Raspian freezes when running my plug in

gbernal, 2018-08-13:
Hi,

I'm wondering if you could help me figure out what is wrong with my plugin. 
after running for a few moment raspian freezes. 

Thank you!

my code is below.

```

from gtts import gTTS
import time
import alsaaudio
import sys
import os

sys.path.append(os.path.abspath("/home/pi/Documents/AH_project/Player/"))
import mp3player 
# Default plugin for output of audio control from environment sound

# print alsaaudio.cards()
# print alsaaudio.Mixer.cardname()

#m = alsaaudio.Mixer('Speaker', cardindex=1)
# m = alsaaudio.Mixer('PCM', cardindex=1)
master, slave = os.openpty()
mixer = alsaaudio.Mixer('Speaker', cardindex=1)
is_mute = mixer.getmute()[0]
usr_vol = mixer.getvolume()

# Call a function here for bike_bell, horn and announcement

def speak(name):
    tts = gTTS(text= name, lang='en')
    tts.save("speaker.mp3")
    os.system("mpg123 speaker.mp3")

def conversation_vol():
    #mixer.setvolume(int(3))
    volume_down()
    print 'someone is calling you'

def volume_up(usr_vol):
    #mixer = alsaaudio.Mixer('Speaker', cardindex=1)
    mixer.setvolume(int(usr_vol))#brings back the audio to the users originally set audio volume

def volume_down():
    
    #mixer = alsaaudio.Mixer('Speaker', cardindex=1)
    mixer.setvolume(int(3)) #this is the lowest volume that alsaaudio can go with the usb audio card

def toggle_mute():
    global is_mute
    mixer.setmute(1 - is_mute)
    is_mute = mixer.getmute()[0]
    if is_mute:
        print('Muted')
    else:
        print('Un-muted')
        
        
def run(readable_results, data, rawbuf):
    #mix = alsaaudio.Mixer('Speaker', cardindex=1)
    global usr_vol

    try:
        if not readable_results:
            print 'empty'
        elif (readable_results[0] == 'hey'):
            conversation_vol()
            time.sleep(10)
            volume_up(usr_vol)
        
        elif (readable_results[0] == 'bike_bell'):
            mp3player.pause()
            speak("Watch out for the bike")
            volume_down()
            time.sleep(5)
            mp3player.pause()
            volume_up(usr_vol)

        elif (readable_results[0] == 'horn'):
            mp3player.pause()
            speak("Watch out for the car")
            volume_down()
            time.sleep(5)
            mp3player.pause()
            volume_up(usr_vol)

        elif ( readable_results[0] == 'announcement' ):
            mp3player.pause()
            speak("announcement")
            volume_down()
            time.sleep(5)
            mp3player.pause()
            volume_up(usr_vol)
        else:
            print "Listening"
    except Exception as err:

        print err

        
        
mp3player.play()

```

mp3player code

```
import os
import subprocess

master, slave = os.openpty()

def play():
    #add function to play song 
    global master
    global mp3_files
    subprocess.Popen(['mpg123', '-C','-q', '/home/pi/Documents/AH_project/Player/track04.mp3'], stdin=master)

    
def stop():
    #add function to quit song
    global slave
    os.write(slave, 'q')
    
    
def pause():
    #add function to pause song 
    global slave
    os.write(slave, 's') 
```

bishoph, 2018-08-14:
You block the whole executing because you use 

<pre>
time.sleep(T)
</pre>

and 

<pre>
subprocess.Popen
</pre>

This means a third of SOPARE can't work anymore for the sleep time and stuff to be executed piles up.

1) Don't block execution with sleep
2) Don't block execution with subprocess (everything waits for return codes and communication)
3) Do use threading or even multiprocessing to deal with long running stuff

closed 2018-08-14 by bishoph

gbernal, 2018-08-14:
@bishoph  Thank you for the feedback, how do you suggest to deal with those type of events when working with sopare?

Thanks

bishoph, 2018-08-14:
You are welcome. I don't understand your question as I made my suggestions in my comment above ;)

gbernal, 2018-08-14:
Thank you. it was too early over here and I read it as Don't use threading ...