bishoph/sopare #37: Broken imports in test/test_audio.py

sllew1024, 2018-09-20:
Following the instructions for initial testing found on the blog and in the README
```
python sopare.py -u
python test/test_audio.py 
```
python throws an error about `import sopare.log`, saying sopare.log coudn't be found. The path is modified to include "../sopare" via `sys.path.append('../sopare')` which means `import log` ought to work (since log is inside the sopare folder), but I found that it didn't (I'm befuddled). Using `sys.path.append('../')` in combination with `import sopare.log` also failed; equally befuddling.

In the interest of making as few changes as possible, I added the main directory to the system path using `sys.path.append(sys.path[0] + '/..')`, thus fixing the imports `sopare.log`, `sopare.config`, and `sopare.audio_factory`. 

I'm not sure whether other people experience this problem -- I'm running ubuntu 18.04 with python 2.7 (the default, I presume). This is my first issue submission, so I hope I've done it all right.

bishoph, 2018-09-20:
What version are you running and how does your directory structure looks like? Please provide the complete command lines after starting incl. the error messages.

I have no idea why you get the error but my gut feeling tells me that your solution is not for the majority of users...

sllew1024, 2018-09-21:
Attached are two text files. error_message.txt details the error message (it's pretty standard for python) as well as some of the debugging I went though to find my solution; directory.txt shows my directory structure (it's exactly the same as the current branch, as I merely git cloned it).

Have you tried running the commands on a machine with the master branch's directory structure? I would be astounded if it worked on anyone's machine in its current state, but I'm certainly no expert on python.

[error_message.txt](https://github.com/bishoph/sopare/files/2403584/error_message.txt)
[directory.txt](https://github.com/bishoph/sopare/files/2403585/directory.txt)

Also, I haven't looked yet, but it occurs to me that there may be similar bugs (or "bugs") in the other test scripts

bishoph, 2018-09-23:
Please provide version information for SOPARE and Python...

sllew1024, 2018-09-27:
My mistake, I was incorrect with my first diagnosis. The real issue is the name of the root directory. If you download the code as a zip, it comes out as sopare-master; if you git clone it, it comes out as sopare. By changing the name from sopare to sopare-master, you can break the imports. The attached text file shows the terminal output for me doing this.

As mentioned above, I've been using the latest sopare and my python is 2.7 and I'm running Ubuntu.

[directoryName.txt](https://github.com/bishoph/sopare/files/2422505/directoryName.txt)

bishoph, 2018-09-28:
Sure, the import for the lib "sopare" must be places and located in the directory named "sopare".

Closing issue.

closed 2018-09-28 by bishoph

sllew1024, 2018-09-28:
I personally would make sure that the code worked out of the box whether cloned via git or downloaded as a zip, but I guess that's up to you to decide.

bishoph, 2018-09-29:
Sure. I currently have no idea how to make sure that the zip works out of the box other than move the whole repository one level up in a newly created directory called sopare...

If anybody has a tip how to solve this easily please comment here :)