bishoph/sopare #38: intermittent threshold change

classicryder, 2018-09-20:
Hi,

I notice that sometimes the threshold value suggestion is changing (not always) after the pi is rebooted, i can assure you that the background noise level is exactly on the same level..

All i can do if this is happening, is keep rebooting the pi until it behaves normally again (the threshold would change back to 100-300, and again in the exactly same background noise level)

Is there any fix for this?

Thanks,
Alvin

bishoph, 2018-09-23:
The threshold is calculated live and at the time you start the test. This means if there is a slightly higher  noise  (even not audible for the human ear) you get higher values. I'm pretty sure this has nothing to do with a restart but only with the environmental sound circumstances :)

hwgeorg, 2018-09-24:
Hi,
danke das war der Fehler. Mein Mikrofon war default auf zu geringen level eingestellt, das hatte ich nicht bemerkt. durch Alvin habe das geprueft, und es ist jetzt ok, also kein Software Fehler (sorry) . Jetzt werde ich lernen und meine Anwendungen pruefen.
gruss hwgeorg

classicryder, 2018-09-26:
The problem is, when the threshold is going high, its really hard to trigger the output (i trigger it by clapping my hand), even if i clap so hard, most of the time it wouldn't be recognized.. in other words : when the threshold goes high, there's nothing i can do.. so i don't think the dynamic threshold can overcome this :)

bishoph, 2018-09-26:
I don't get it. You define the threshold once. Make sure to define and use a (low enough) value that triggers always when you clap your hand. No need to change the threshold if you have the right value. Maybe you need to describe your steps and tell me what you do when and how to make it clear :)

classicryder, 2018-09-27:
Ok heres what i did:

1. I use sopare to toggle a lamp by clapping my hand, so i expect the clap wouldn't be much harder/louder than it should be for the sake of convenience.
2. I set the threshold to 250-300 (so the clap wouldn't need to be so hard/loud enough and still complying the threshold recommendation from test_audio command)
3. I train the clap, and the other sounds
4. Compile it all to dict.json
5. Use it daily
6. But sometimes i notice my clap cant get recognized at all, so i kill the sopare and run the test_audio again.. the threshold recommendation goes really high (4000 or more), at this point, i can change the threshold to 4000 or more but i have to clap so loud and hard, its not convenient for me, and still sometimes the clap cant get recognized (i believe this is because the clap isnt loud enough)

I hope this clarify the chronology :)

classicryder, 2018-09-27:
And if i set the threshold low in order to recognize the clap on whatever the threshold recommendation is, then the clap would be so hard to be recognized because : the threshold is too low, so any kind of noises are being picked up, hence the time i clap my hand might overlap with the other noises thats being picked up.. i can see the noises being picked up because when I watch it in terminal, even when i don't clap my hand and not making any noise, the [] keeps on showing (and sometimes other trained sound like ['chair'], ['knock'], etc)

bishoph, 2018-09-28:
The recommendation is just a recommendation and should be treated like such. If you run the audio test with music in the background you get a high value recommendation, In your case you have a quite environment  and over the day something happens and it gets nosier and louder. You should use the lowest threshold that triggers SOPARE and detects your sound when you train at several times. Not sure if this is capable but you can give it a try...

classicryder, 2018-09-28:
btw this is the config that i use: the chunk =1000, sample rate=44100 and chunks=4000 are intentionally used to overcome the Input overflow issue (the very first problem that i encountered)

#########################################################
# Stream prep and silence configuration options #########
#########################################################

[stream]

# Read chunk size
CHUNK = 1000

# Sample rate
SAMPLE_RATE = 44100

# Volume threshold when audio processing starts / silence
THRESHOLD = 330

# Silence time in seconds when analysis is called
MAX_SILENCE_AFTER_START = 0.5

# Time in seconds after the analysis is forced
MAX_TIME = 2.4

# Start the analysis after reaching LONG_SILENCE
LONG_SILENCE = 15

# Characteristic length
CHUNKS = 4000

#########################################################
# Characteristic configuration options ##################
#########################################################

[characteristic]

# Steps boil down the data into smaller chunks of data.
# Smaller steps mean more precision but require
# normally more learned entries in the dictionary.
# Progressive value is used if you want to pack not
# so relevant frequencies
PROGRESSIVE_FACTOR = 0
START_PROGRESSIVE_FACTOR = 600
MIN_PROGRESSIVE_STEP = 5
MAX_PROGRESSIVE_STEP = 5

# Specifies freq ranges that are kept for further
# analysis. Freq outside of the ranges are set to zero.
# Human language can be found between 20 and 5000.
LOW_FREQ = 20
HIGH_FREQ = 600

# Make use of Hann window function
HANNING = true

# Range factor for peaks
PEAK_FACTOR = 0.7

#########################################################
# Compare configuration options #########################
#########################################################

[compare]

# Min. number of tokens to identify the beginning of a word
MIN_START_TOKENS = 3

# Min. value for potential beginning of a word
MARGINAL_VALUE = 0.4

# Minimal similarity across all comparison to
# identify a complete word across all tokens
#changed
MIN_CROSS_SIMILARITY = 0.8

# Calculation basis or token/word comparison
SIMILARITY_NORM = 0.6
SIMILARITY_HEIGHT = 0.4
SIMILARITY_DOMINANT_FREQUENCY = 0

# Number of best matches to consider.
# Value must be > 0
# If not specified or value < 1 value is set to 1
#changed
NUMBER_OF_BEST_MATCHES = 2

# Min. distance to keep a word
MIN_LEFT_DISTANCE = 0.9
MIN_RIGHT_DISTANCE = 0.7

# Use given number as results to assembly result
# 0 for all predictions
MAX_WORD_START_RESULTS = 2
MAX_TOP_RESULTS = 3

# Enable or disable strict length check for words
STRICT_LENGTH_CHECK = true
# Value to soften the strict length check a bit to still
# get quite precise results but to be less strict
STRICT_LENGTH_UNDERMINING = 2

# Short term memory retention time in seconds. Zero to disable STM
STM_RETENTION = 1.2

# Fill result percentage
# 0.5 means that half of the values can by empty to still get valid results
# A lower value should theoretically avoid false positives
FILL_RESULT_PERCENTAGE = 0.1

#########################################################
# Misc configuration options ############################
#########################################################

[misc]

# Loglevel (CRITICAL, ERROR, WARNING, INFO, DEBUG)
LOGLEVEL = ERROR

#########################################################
# Experimental configuration options ####################
#########################################################

[experimental]

# Additional FFT analysis and comparison for CHUNKS/2 length
FFT_SHIFT = false

bishoph, 2018-09-28:
Completely silent would give you a recommendation between zero and 100 :)

What you can do is to visualize the current "silence". Try the plot option and see whats going on.

classicryder, 2018-09-29:
So today i use a new usb sound adapter and it works.. maybe its a hardware problem all this time.. l'll monitor it for a week or two, then ill close this issue :)

For the plot feature, could you please add the label on the x and y axis? I couldn't understand the meaning of the chart :D

bishoph, 2018-09-29:
Ok, please report back if the new mic solves the issue. I mark the bug as enhancement for the axis labeling :)

classicryder, 2018-10-08:
Looks like the new usb mic port solved the issue 

Thanks a lot for the help

bishoph, 2018-10-23:
You are welcome and a big thank you for the feedback and your report back. Never thought this could be a hardware issue.

closed 2018-10-23 by bishoph