bishoph/sopare #42: Could not recognize i2s MEMS microphone

xastx, 2018-11-21:
Hi,
I'm able to track environmental noises and detect certain sounds from my pc. I also have a pi and i2s mems microphone, which record sounds with arecord command in wav format smoothly. However when I try to run the program from my pi, my recommended threshold is too high (around 8k) and also it gives me an error about samplerate matching. I guess it recognized my microphone but something went wrong and I'm not able to use it because it does not sense any sounds. Should I do any more settings

xastx, 2018-11-22:
Thanks for reply, my results for the setup is in below. Can you check them?

```
pi@raspberrypi:~/dev/sopare $ ./sopare.py -v -t test 
sopare 1.5.0
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
connect(2) call to /tmp/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
INFO:sopare.analyze:checking for plugins...
DEBUG:sopare.analyze:loading and initialzing plugins/print
DEBUG:sopare.audio_factory:#### Default input device info #####
DEBUG:sopare.audio_factory:defaultSampleRate: 44100.0
DEBUG:sopare.audio_factory:defaultLowOutputLatency: 0.0087074829932
INFO:sopare.worker:worker queue runner started
DEBUG:sopare.audio_factory:defaultLowInputLatency: 0.00580498866213
DEBUG:sopare.audio_factory:maxInputChannels: 2
DEBUG:sopare.audio_factory:structVersion: 2
DEBUG:sopare.audio_factory:hostApi: 0
DEBUG:sopare.audio_factory:index: 2
DEBUG:sopare.audio_factory:defaultHighOutputLatency: 0.0348299319728
DEBUG:sopare.audio_factory:maxOutputChannels: 2
DEBUG:sopare.audio_factory:name: snd_rpi_simple_card: - (hw:1,0)
DEBUG:sopare.audio_factory:defaultHighInputLatency: 0.0348299319728
DEBUG:sopare.recorder:SAMPLE_RATE: 48000
INFO:sopare.buffering:buffering queue runner
DEBUG:sopare.recorder:CHUNK: 512
INFO:sopare.recorder:start endless recording
```
```
pi@raspberrypi:~/dev/sopare $ python test/test_audio.py
test_audio init...
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
[... 56 lines cut ...]
Excellent. Got all 40960 chunks.
Excellent. Got all 81920 chunks.
Excellent. Got all 5120 chunks.
Excellent. Got all 10240 chunks.
Excellent. Got all 20480 chunks.
Excellent. Got all 40960 chunks.
Excellent. Got all 81920 chunks.
Excellent. Got all 5120 chunks.
Excellent. Got all 10240 chunks.
Excellent. Got all 20480 chunks.
Excellent. Got all 40960 chunks.
Excellent. Got all 81920 chunks.
Excellent. Got all 5120 chunks.
Excellent. Got all 10240 chunks.
Excellent. Got all 20480 chunks.
Excellent. Got all 40960 chunks.
Excellent. Got all 81920 chunks.

Your sopare/config.py recommendations:

SAMPLE_RATE = 48000
CHUNK = 512
THRESHOLD = 8200

```

bishoph, 2018-11-22:
Seems be ok so far. Your THRESHOLD recommendation is very high. This could be due to noise while testing. Lower the THRESHOLD  in your settings to a value around 500 and try to record, train and run SOPARE. If this fails your soundcard records lots of noise and there is not so much I can do in that case....