bishoph/sopare #47: Recognize with a file

Mario0606, 2019-01-03:
Hello and happy new year, I would like to know if there is any way I can recognize the voice through a file instead of being by the microphone (loop)? I know I can use files to train, but can I use files to recognize them?

bishoph, 2019-01-03:
Sure, as documented. You can write and read files. With the files you can run normal operations like training or recognition. I use these features for testing and to get reproduceable results.

Mario0606, 2019-01-07:
Hello, I was able to do the training and the test through files. Contextualizing, I'm doing a college activity where I want the program's response to be just "yes" or "no", I'm training through four different people's audio files and also testing through files, but the accuracy is very low , I would like to know if you have had some similar experience and where I would probably be wrong, if it is something in default.ini, or if the precision in training with files is smaller than directly from the microphone, or even if you suspect be my database.

bishoph, 2019-01-07:
As I no nothing about your training nor your configuration I can only guess. For many different voices 4 people seems to be ok. Mind that the file input may has data upfront and at the end that may change the results. Therefor precise recording is key! View your input and cross check the data with the plot option. Config wise I would try to focus on the dominant frequency and add SIMILARITY_NORM if needed. Later on decrease distances to avoid false positives. You will still get many as yes and no are very short words. not sure if the length check must be enabled or disabled. Requires testing and adjusting.

chuddy1, 2022-10-11:
Hi how exactly can I run the recognition through files? Would it read a wav file for recognition or a raw file? thank you

bishoph, 2022-10-11:
I added recognition through files for debugging and testing purpose. Only raw files are supported. The options are:

<code>
 -w --write  [file]  : write raw to [dir/filename]

 -r --read   [file]  : read raw from [dir/filename]
</code>

chuddy1, 2022-10-12:
Hi thanks for your reply. I tried to use the write function but it does not allow me to write files for a duration longer than 5 seconds. I am trying to record up to 10mins sound so that I could perform sensitivity analysis on the characteristics which I have used in Sopare. I am trying to get the best possible combination of characteristics I need to detect my key word. I also tried reading a raw file which contains my keyword in it but Sopare did not detect it. As a matter of fact it analyzes every raw file I read as not containing my keyword. I am not sure if I am doing this correctly.

Thank you very much for your time!

bishoph, 2022-10-13:
The name "raw file" is misleading as the content is processed by SOPARE and modified based on the config parameters. This means the length of the raw file is determined by the config parameters like MAX_TIME etc. One conclusion of this is that raw files are heavily based on a specific config. As soon as you change the config you should delete the raw files and create new ones or store raw files config specific. Hope that clarifies the topic.

chuddy1, 2022-10-24:
Thank you bishoph for the reply. I understand now.

chuddy1, 2022-10-24:
I also made an observation when running tests with my Sopare. I have 5 Raspberry Pis which are the exact clones of themselves. I used Sopare on all 5 Raspberries within the same environment but I got different predication results between all 5 Raspberries. The furthest distance between 2 Raspberries in my setup is 29 inches. I am not sure why I am getting different prediction results. Is this an expected behavior from the software?

bishoph, 2022-10-25:
Depends on the environment. Echos, obstructions and even the volume can vary for the same sound in a room for different machines a lot. Try to analyse one sound on all 5 Pis and you may find the differences even in the graphical  representation easily.