bishoph/sopare #50: Issues setting up on OS X

pantoninho, 2019-02-26:
I'm having some problems setting up Sopare on OSX Sierra.
After some investigation, I've found out that there are some issues with the multiprocessing module on osx.
https://docs.python.org/2/library/multiprocessing.html#multiprocessing.Queue.qsize
>>>
qsize()
Return the approximate size of the queue. Because of multithreading/multiprocessing semantics, this number is not reliable.

Note that this may raise NotImplementedError on Unix platforms like Mac OS X where sem_getvalue() is not implemented.
>>>
also:
https://github.com/vterron/lemon/issues/11

Since I'm completely alien to the python ecosystem, I'm not sure if there is a known workaround for this problem (I did not find any).
Can you point me in the right direction?

Thanks.

Error while running `python ./test/test_audio.py`
```
Traceback (most recent call last):
  File "test/test_audio.py", line 132, in <module>
    ta.stop()
  File "test/test_audio.py", line 118, in stop
    while (self.queue.qsize() > 0):
  File "/usr/local/Cellar/python@2/2.7.15_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/multiprocessing/queues.py", line 143, in qsize
    return self._maxsize - self._sem._semlock._get_value()
NotImplementedError
```

Error while running `python sopare.py -t "test"`
```
sopare 1.5.0
Traceback (most recent call last):
  File "/usr/local/Cellar/python@2/2.7.15_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/multiprocessing/queues.py", line 268, in _feed
    send(obj)
IOError: [Errno 32] Broken pipe
```

bishoph, 2019-02-26:
According to the documentation the "NotImplementedError" on Mac OS X systems is still inherent. To avoid this error one could simply change the code and check if the queue is full instead qsize() > 0
See https://docs.python.org/3.4/library/multiprocessing.html?highlight=process for details.

The other error message is also multiprocessing related. It happens while SOPARE shuts down and one process still works with the queue. Maybe you killed one process manually? AFAIR there is some code to avoid this but I have to cross check ;)

pantoninho, 2019-03-04:
Thanks for the quick reply, @bishoph!

Your suggestion to replace `queue.qsize() > 0` with `queue.full()` fixed the `test_audio.py` script error, but I'm still not able to pass through the second problem (the error while running `sopare.py` script).

I've made sure only one sopare process is running, with no success.

bishoph, 2019-03-31:
Fixed with commit ec13b29 in testing branch (version >= 1.5.4)

closed 2019-03-31 by bishoph