bishoph/sopare #51: ascii' codec can't encode character  error on PI 3 B+

NewToPi, 2019-03-09:
[PostedIssues.txt](https://github.com/bishoph/sopare/files/2948014/PostedIssues.txt)

bishoph, 2019-03-09:
Thanks for the report.
Need to use proper utf-8 encoding before using str method as the console encoding is different to utf-8.

Untested fix:
self.logger.debug(str(k.encode('utf-8')) + ': ' + str(v.encode('utf-8')))

NewToPi, 2019-03-09:
Thank you Sir 
I am assuming I need to replace the self logger code with this code , as I am new at this can you tell me exactly how I need to change this. I apologize for my lack of knowledge. Again I thank you for your helping me.

bishoph, 2019-03-09:
Exactly, you can replace the line by yourself and save the file with your favorite editor. As I don't know much about your environment (and I don't want you to last in vi forever) my recommendation is to use the tools you are familiar with. Or you simply wait until I fix it ;)

bishoph, 2019-03-31:
Should be fixed with commit ac33893 in testing branch

closed 2019-03-31 by bishoph