bishoph/sopare #52: Sending Curl command from python script

onacvooe, 2019-04-01:
Hi,

Is it possible to execute commands from the python script?
As an example:

`def run(readable_results, data, rawbuf):
    if ('lights' in readable_results and 'on' in readable_results):
       execute (curl command)` #Or any other shell action for that matter

bishoph, 2019-04-01:
Here is a link that should get you going:

https://stackoverflow.com/questions/25491090/how-to-use-python-to-execute-a-curl-command

onacvooe, 2019-04-01:
> Here is a link that should get you going:
> 
> https://stackoverflow.com/questions/25491090/how-to-use-python-to-execute-a-curl-command

I have tried different options mentioned with searching on google but I can't get anything to work. Sometimes sopare doesn't start at all with a message saying a syntax error or something. Other times it starts fine and executes all given commands but my lights don't respond and I don't get a error message. Any help would be great.

Thanks for all your work and help.

onacvooe, 2019-04-01:
My current configuration is doesn't generate any errors but as you can see I put in your original message "Tschakka! Got my two words...now do some awesome stuff" at the end, so I know it runs the other command as well.

I tried executing `./sopare -v -l` for verbose output, which doesn't help, and also normal `./sopare -l`
And it outputs the following:

`lappie@lappie01:~/sopare$ ./sopare.py -l
sopare 1.5.0
ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to open slave
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
Cannot lock down 82280346 byte memory area (Cannot allocate memory)
[u'lampen']
Tschakka! Got my two words...now do some awesome stuff
`

When running with the verbose in it, I get the following output:

lappie@lappie01:~/sopare$ ./sopare.py -l -v
sopare 1.5.0
ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to open slave
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
Cannot lock down 82280346 byte memory area (Cannot allocate memory)
INFO:sopare.analyze:checking for plugins...
DEBUG:sopare.analyze:loading and initialzing plugins/print
DEBUG:sopare.analyze:loading and initialzing plugins/lampen01
INFO:sopare.worker:worker queue runner started
DEBUG:sopare.audio_factory:#### Default input device info #####
DEBUG:sopare.audio_factory:defaultSampleRate: 44100.0
DEBUG:sopare.audio_factory:defaultLowOutputLatency: 0.0087074829932
DEBUG:sopare.audio_factory:defaultLowInputLatency: 0.0087074829932
DEBUG:sopare.audio_factory:maxInputChannels: 32
DEBUG:sopare.audio_factory:structVersion: 2
DEBUG:sopare.audio_factory:hostApi: 0
DEBUG:sopare.audio_factory:index: 4
DEBUG:sopare.audio_factory:defaultHighOutputLatency: 0.0348299319728
DEBUG:sopare.audio_factory:maxOutputChannels: 32
DEBUG:sopare.audio_factory:name: default
DEBUG:sopare.audio_factory:defaultHighInputLatency: 0.0348299319728
INFO:sopare.buffering:buffering queue runner
DEBUG:sopare.recorder:SAMPLE_RATE: 48000
DEBUG:sopare.recorder:CHUNK: 512
INFO:sopare.recorder:start endless recording
INFO:sopare.processing:starting append mode
DEBUG:sopare.filter:New window!
DEBUG:sopare.worker:characteristic = 0 {'dfm': 256784, 'peaks': [0, 1, 2, 3, 4, 5, 6, 7, 8, 31, 32], 'df': 21, 'volume': 488, 'fc': 208.3, 'norm': [0.24748985259337808, 0.19697923196415063, 0.17613148

**[[## -= CUT OUT A LOT OF NUMBERS FOR CLARITY =- ##]]** 

0.06052299471687352], 'token_peaks': [18020, 59187, 149084, 176322, 198786, 206417]}
DEBUG:sopare.worker:meta = [{'adapting': 206417, 'token_peaks': [18020, 59187, 149084, 176322, 198786, 206417], 'pos': 6, 'volume': 488, 'token': 'token', 'silence': 0}]
DEBUG:sopare.worker:characteristic = 1 {'dfm': 1080438, 'peaks': [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 11, 37, 39, 40, 41, 42], 'df': 37, 'volume': 1468, 'fc': 185.4, 'norm': [0.18862625101780392, 0.16651922341725578, 0.18209453292238723, 0.20219862790041623, 0.19165117417294977, 

**[[## -= CUT OUT A LOT OF NUMBERS FOR CLARITY =- ##]]** 

0.05042740354542106, 0.06486819269188564, 0.053484397538042544], 'token_peaks': [218664, 263590, 322222, 463373, 697530, 590549]}
DEBUG:sopare.worker:meta = [{'adapting': 590549, 'token_peaks': [218664, 263590, 322222, 463373, 697530, 590549], 'pos': 12, 'volume': 1468, 'token': 'token', 'silence': 0}]
DEBUG:sopare.worker:characteristic = 2 {'dfm': 1336807, 'peaks': [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 36, 37, 38, 39, 40, 41], 'df': 38, 'volume': 639, 'fc': 205.6, 'norm': [0.18419380846494518, 0.1541841647025054, 0.19131611625405495, 0.20266483341287808, 0.18899637205132147, 

**[[## -= CUT OUT A LOT OF NUMBERS FOR CLARITY =- ##]]**

0.07964203724066227, 0.06843676819066119, 0.07589014211555282, 0.05920830241633842, 0.05369450300744336, 0.05894261875232369, 0.052368810603330436], 'token_peaks': [682449, 712910, 581495, 530845, 297026, 258813]}
DEBUG:sopare.worker:meta = [{'adapting': 258813, 'token_peaks': [682449, 712910, 581495, 530845, 297026, 258813], 'pos': 18, 'volume': 639, 'token': 'token', 'silence': 0}]
DEBUG:sopare.worker:characteristic = 3 {'dfm': 125066, 'peaks': [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 27, 41], 'df': 25, 'volume': 271, 'fc': 205.2, 'norm': [0.17719482163526487, 0.16943697186227125, 0.16524615616689897, 0.16819665659339972, 0.14704891173808068, 0.16237724936352846, 

**[[## -= CUT OUT A LOT OF NUMBERS FOR CLARITY =- ##]]**

0.05855445108328623, 0.06576568136339754, 0.05935504535502074, 0.05656116049371686, 0.06277894858830546, 0.06309620694653208], 'token_peaks': [287068, 186861, 199308, 137319, 113162, 124920]}
DEBUG:sopare.worker:meta = [{'adapting': 124920, 'token_peaks': [287068, 186861, 199308, 137319, 113162, 124920], 'pos': 24, 'volume': 271, 'token': 'token', 'silence': 3}]
DEBUG:sopare.worker:characteristic = 4 {'dfm': 378059, 'peaks': [0, 1, 2, 3, 4, 5, 6, 8], 'df': 32, 'volume': 977, 'fc': 197.5, 'norm': [0.1531105318877866, 0.14708024560636948, 0.16810067633547227, 0.14123896330526972, 0.14287343968811164, 0.13803642298467456, 
[... 129 lines cut ...]
DEBUG:sopare.analyze:removing lampen from potential start position 0 bc MARGINAL_VALUE > 0
DEBUG:sopare.analyze:************************************************

[({'dfm': 7648, 'peaks': [1, 2, 3, 4, 5, 7], 'df': 36, 'volume': 0, 'fc': 183.4, 'norm': [0.11774434726189205, 0.15434745135066458, 0.16816839827022176, 0.17350093663669866, 0.14618738294625594, 

[... cut ...]
0.08728865309239563, 0.08920844339605598, 0.08635693329399861, 0.08707555505627554, 0.09642622374704964, 0.08758294494736861, 0.0800310549151561], 'token_peaks': []}, [{'adapting': 0, 'peaks': [10907, 8136, 8171], 'pos': 3, 'volume': 0, 'token': 'start analysis', 'silence': 131}])]

{u'aan': [[0.2732126859123079]], u'lampen': [[0.2986424257165124]]}

framing_match: []

sorted_best_match: []

match_results: ['']

Results contain too many empty tokens. 1 / 1 Eliminating results`
`

When putting the Curl command in my terminal it works just fine.

Maybe I'm missing something?

Thanks for looking at this.

bishoph, 2019-04-01:
I can help in terms of SOPARE but not in terms of "my custom plugin curl command is not working and I don't know why". I suggest that you get SOPARE going without custom code. By going I mean it should run well until you get reproducible results. After you reached this goal start step two and throw in your custom plugin code and debug until the plugin code works. It seems to me that you have two loose ends. 

It may help if you write a short Python program that runs standalone to test the execution of the curl command and the communication at both ends.

Verbose mode is fine for debugging SOPARE but not to find errors in custom plugins BTW ;)

onacvooe, 2019-04-01:
> I can help in terms of SOPARE but not in terms of "my custom plugin curl command is not working and I don't know why". I suggest that you get SOPARE going without custom code. By going I mean it [...]

I know, SOPARE runs fine, without the custom plugin. I'm really not that good in understanding Python tho. I'll see what I can do with writing a Python program. But I'm more the visual kind of guy ;-)

Thanks anyway for your help! I'll post my sollution here, if I find any...

onacvooe, 2019-04-01:
After a lot of trial and error, the following code works (for me) and I would like to share it here in case someone needs it:

`#!/usr/bin/env python
# -*- coding: utf-8 -*-

"""
Copyright (C) 2015 - 2017 Martin Kauss (yo@bishoph.org)

Licensed under the Apache License, Version 2.0 (the "License"); you may
not use this file except in compliance with the License. You may obtain
a copy of the License at

 http://www.apache.org/licenses/LICENSE-2.0

Unless required by applicable law or agreed to in writing, software
distributed under the License is distributed on an "AS IS" BASIS, WITHOUT
WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied. See the
License for the specific language governing permissions and limitations
under the License.
"""

import subprocess

# Default plugin for output of analysis
def run(readable_results, data, rawbuf):
    if ('lampen' in readable_results):
	subprocess.call("curl -H 'Content-Type: text/plain' -X POST -d 'OFF' http://IPADRESS:PORT/rest/items/wcd001_schakelaar", shell=True)`

closed 2019-04-02 by onacvooe