bishoph/sopare #53: Running all options in __init__.py

onacvooe, 2019-04-02:
Hi,
I suddenly have the following problem, when starting Sopare and give a voicecommand, Sopare runs all commands inside plugins folder. Even when it doesn't recognize any word.
When it doesn't recognize a word the output shows this:

`[]
Lampen zijn aan
Lampen zijn aan
Lampen zijn uit
Lampen zijn uit
[]
`
and nothings happens (no lights go on/off).

When it detects a word, it executes the commands behind it (both commands)
So when I'm saying a trained word like "Lights On" and sopare recognizes it, it starts at the top of the init.py file and turns my lights on, they then immediately turn off. Because the next command is the lights off command.

If I speak the lights off command, he first runs the light on command and then the light off command.

Here is my setup....

my config/default.ini : 

`#########################################################
# Stream prep and silence configuration options #########
#########################################################

[stream]

# Read chunk size
CHUNK = 512

# Sample rate
SAMPLE_RATE = 48000

# Volume threshold when audio processing starts / silence
THRESHOLD = 350

# Silence time in seconds when analysis is called
MAX_SILENCE_AFTER_START = 1.4

# Time in seconds after the analysis is forced
MAX_TIME = 2.4

# Start the analysis after reaching LONG_SILENCE
LONG_SILENCE = 20

# Characteristic length
CHUNKS = 3072

#########################################################
# Characteristic configuration options ##################
#########################################################

[characteristic]

# Steps boil down the data into smaller chunks of data.
# Smaller steps mean more precision but require
# normally more learned entries in the dictionary.
# Progressive value is used if you want to pack not
# so relevant frequencies
PROGRESSIVE_FACTOR = 0
START_PROGRESSIVE_FACTOR = 600
MIN_PROGRESSIVE_STEP = 5
MAX_PROGRESSIVE_STEP = 5

# Specifies freq ranges that are kept for further
# analysis. Freq outside of the ranges are set to zero.
# Human language can be found between 20 and 5000.
LOW_FREQ = 20
HIGH_FREQ = 600

# Make use of Hann window function
HANNING = true

# Range factor for peaks
PEAK_FACTOR = 0.7

#########################################################
# Compare configuration options #########################
#########################################################

[compare]

# Min. number of tokens to identify the beginning of a word
MIN_START_TOKENS = 3

# Min. value for potential beginning of a word
MARGINAL_VALUE = 0.7

# Minimal similarity across all comparison to
# identify a complete word across all tokens
MIN_CROSS_SIMILARITY = 0.8

# Calculation basis or token/word comparison
SIMILARITY_NORM = 0.6
SIMILARITY_HEIGHT = 0.4
SIMILARITY_DOMINANT_FREQUENCY = 0

# Number of best matches to consider.
# Value must be > 0
# If not specified or value < 1 value is set to 1
NUMBER_OF_BEST_MATCHES = 2

# Min. distance to keep a word
MIN_LEFT_DISTANCE = 0.9
MIN_RIGHT_DISTANCE = 0.7

# Use given number as results to assembly result
# 0 for all predictions
MAX_WORD_START_RESULTS = 2
MAX_TOP_RESULTS = 3

# Enable or disable strict length check for words
STRICT_LENGTH_CHECK = true
# Value to soften the strict length check a bit to still
# get quite precise results but to be less strict
STRICT_LENGTH_UNDERMINING = 2

# Short term memory retention time in seconds. Zero to disable STM
STM_RETENTION = 1.2

# Fill result percentage
# 0.5 means that half of the values can by empty to still get valid results
# A lower value should theoretically avoid false positives
FILL_RESULT_PERCENTAGE = 0.1

#########################################################
# Misc configuration options ############################
#########################################################

[misc]

# Loglevel (CRITICAL, ERROR, WARNING, INFO, DEBUG)
LOGLEVEL = ERROR

#########################################################
# Experimental configuration options ####################
#########################################################

[experimental]

# Additional FFT analysis and comparison for CHUNKS/2 length
FFT_SHIFT = false
`

My __init__.py

#!/usr/bin/env python
# -*- coding: utf-8 -*-

"""
Copyright (C) 2015 - 2017 Martin Kauss (yo@bishoph.org)

Licensed under the Apache License, Version 2.0 (the "License"); you may
not use this file except in compliance with the License. You may obtain
a copy of the License at

 http://www.apache.org/licenses/LICENSE-2.0

Unless required by applicable law or agreed to in writing, software
distributed under the License is distributed on an "AS IS" BASIS, WITHOUT
WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied. See the
License for the specific language governing permissions and limitations
under the License.
"""

import subprocess

# Lampen aan - Spraak
def run(readable_results, data, rawbuf):
 if ('dubbelklap' in readable_results):
  subprocess.call("curl -H 'Content-Type: text/plain' -X POST -d 'ON' http://192.168.1.253:8080/rest/items/wcd001_schakelaar", shell=True)
 print ('Lampen zijn aan')
# Lampen aan - Klappen
 if ('lampen-aan' in readable_results):
  subprocess.call("curl -H 'Content-Type: text/plain' -X POST -d 'ON' http://192.168.1.253:8080/rest/items/wcd001_schakelaar", shell=True)
 print ('Lampen zijn aan')

# Lampen uit - Spraak
 if ('lampen-uit' in readable_results):
  subprocess.call("curl -H 'Content-Type: text/plain' -X POST -d 'OFF' http://192.168.1.253:8080/rest/items/wcd001_schakelaar", shell=True)
 print ('Lampen zijn uit')
 if ('vingerknip' in readable_results):
  subprocess.call("curl -H 'Content-Type: text/plain' -X POST -d 'OFF' http://192.168.1.253:8080/rest/items/wcd001_schakelaar", shell=True)
 print ('Lampen zijn uit')

I'm running Ubuntu 18.04

What could be wrong?

Thanks for your help.

bishoph, 2019-04-02:
As SOPARE did not change suddenly my suggestion is you take a look at your changes. If you report bugs and issues please make sure the issue is reproducible. On top of this provide steps how to reproduce. Otherwise submitting new issues makes not a lot of sense as my crystal ball is currently out of order ;)

Hint: If you post code make sure to add &lt;pre&gt; or &lt;code&gt; tags so that the code is readable. Or attach the file.

onacvooe, 2019-04-03:
Thank you, I have re-install it all and now it seems to be working, but I cannot see any difference in the code... So very strange. But it is working again.

closed 2019-04-03 by onacvooe

bishoph, 2019-04-03:
That's the beauty of programming :)

Have fun!