bishoph/sopare #54: Toggling output using sopare and txt file

classicryder, 2019-04-07:
Hi,

Im trying to toggle the led output based on the output of a text file (True or False), so whenever sopare detect 2 consecutive claps, it will open the text file, read it, save the true/false value in a variable called "state", then negate it and send it to the GPIO. I have tried to mod the code so it becomes like this:

def run(readable_results, data, rawbuf):
    time = strftime("%Y-%m-%d, %H:%M:%S %Z", localtime())
    print time, readable_results
    global state
   
    if readable_results==[u'clap']:
     GPIO.output(35, True)          				   #
     sleep(0.025)							   # flash my indicator led if its just 1 clap
     GPIO.output(35, False)                                          #
    
    if readable_results==[u'clap', u'clap']:
     f = open('/home/pi/sopare/plugins/clap/state.txt')
     state = f.read()					 # open the txt file to get the state value then close it
     f.close()						 #
     print "after reading:",state			 #
     state=bool('state')				 #
     f.close()						 #
    
     GPIO.output(35, True)			 #
     sleep(0.025)					 # flash my indicator led
     GPIO.output(35, False)	                 #
     state = not state                                 # negate the state value
     GPIO.output(37, state)                        # toggle the light output
     print "Light is toggled"
     print "after negating:",state

     f = open('/home/pi/sopare/plugins/clap/state.txt', "w") 
     f.write(repr(state))                                   # update the txt file with the latest state (true or false)
     f.close()                                                  

But the strange thing is, it only works on the first cycle of the sound pattern recog, please refer to the screenshot attached

[image]

please help

Thanks
Alvin

classicryder, 2019-04-10:
Hi

I have solved this issue

Thanks

bishoph, 2019-04-14:
Great! Closing the issue for now.

closed 2019-04-14 by bishoph

classicryder, 2019-04-16:
Hi,

so my method is to have another shell script running all the time to check the text file and whenever the text file is changed, the GPIO is toggled, but using that approach causes my raspberry pi to freeze/reboot after 2-3 hours of run.. so im thinking, is there any way to "inject" a simulated saved sound pattern while sopare is running to toggle the gpio? like by pushing a button connected to the GPIO pin?

bishoph, 2019-04-17:
A workaround is never the best solution. Try to fix the root cause.