bishoph/sopare #57: Random Results from Custom Plugin

hankp46, 2019-04-14:
Hello
I wrote this custom plugin as a beginning to a robot project i am working on. 
I am getting some strange results. It recognizes my words and prints correct results but sometimes also prints random results, almost like it remembered words from a previous session (see bottom for some printout)
Is there a file or buffer I should be clearing prior to initiating
(I'm a newbie)
#!/usr/bin/env python
# -*- coding: utf-8 -*-

"""
Copyright (C) 2015 - 2017 Martin Kauss (yo@bishoph.org)

Licensed under the Apache License, Version 2.0 (the "License"); you may
not use this file except in compliance with the License. You may obtain
a copy of the License at

 http://www.apache.org/licenses/LICENSE-2.0

Unless required by applicable law or agreed to in writing, software
distributed under the License is distributed on an "AS IS" BASIS, WITHOUT
WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied. See the
License for the specific language governing permissions and limitations
under the License.
"""

def Forward():
    print("Robocar Moving Forward")

def Back():
    print("Robocar Moving backward")

def Stop():
    print("Robocar Stopping")
    
def run(readable_results, data, rawbuf):
    if (len(readable_results) > 2 or len(readable_results) == 0):
        return
    try:
        if ('forward' in readable_results) :
            Forward()
        elif ('back' in readable_results) :
            Back()
        elif ('stop' in readable_results) :
            Stop()
        
    except Exception as err:
        print (err)
 Results:
 ./sopare.py -l
[u'forward']
Robocar Moving Forward
[]
[u'forward']
Robocar Moving Forward
[]
[]
[]
[]
[u'back']
Robocar Moving backward
[u'back', u'back']
Robocar Moving backward
[]
[]
[u'back']
Robocar Moving backward
[]
[]
[]
[]
[]
[u'stop']
Robocar Stopping
[]
[]
[]
[u'stop']
I never said stop.  Plus why is it printing {u'stop'} or any command when my plugin does not have a
 print readable_results   
statement?

a3020, 2019-04-14:
>  Plus why is it printing {u'stop'} or any command when my plugin does not have a
print readable_results

Maybe you still have the default `print` plugin present? All plugins from the `plugins` directory are loaded, see https://github.com/bishoph/sopare/blob/873242efda8d833cab91cb68c64bcfe9eaec42e0/sopare/analyze.py#L189-L197

bishoph, 2019-04-14:
There is nothing you have to clear. Make sure that your plugin code runs fast and does not slow down the main program. Decouple slow running code by using threads or alike.

On the other hand if your mic is not very precise and your training is rough other sounds can trigger the recognition of trained words.

hankp46, 2019-04-15:
Removing the default plugin solved the problem. Thank you

closed 2019-04-15 by hankp46