bishoph/sopare #58: Plot error

a3020, 2019-04-15:
When using the '-p' option, the program crashes with an error. This happens in the testing and master branch:

> INFO:sopare.recorder:start endless recording
INFO:sopare.processing:starting append mode
/usr/lib/python2.7/dist-packages/numpy/core/numeric.py:531: ComplexWarning: Casting complex values to real discards the imaginary part
  return array(a, dtype, copy=False, order=order)
INFO:sopare.processing:stop append mode because of silence
INFO:sopare.buffering:stop buffering
INFO:sopare.buffering:terminating queue runner
INFO:sopare.recorder:Buffering not alive, stop recording
INFO:sopare.recorder:stop endless recording
Process worker for filtered data:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/home/pi/Documents/sopare/sopare/worker.py", line 123, in run
    self.analyze.do_analysis(self.compare.get_results(), self.character, self.rawbuf)
  File "/home/pi/Documents/sopare/sopare/analyze.py", line 55, in do_analysis
    matches = self.deep_search(framing, data)
  File "/home/pi/Documents/sopare/sopare/analyze.py", line 91, in deep_search
    word_sim = self.deep_inspection(id, startpos, data)
  File "/home/pi/Documents/sopare/sopare/analyze.py", line 137, in deep_inspection
    sim, sl, sr = self.token_sim(characteristic, dcharacteristic)
  File "/home/pi/Documents/sopare/sopare/analyze.py", line 120, in token_sim
    sim_zcr = self.util.single_similarity(characteristic['zcr'], dcharacteristic['zcr']) * self.cfg.getfloatoption('compare', 'SIMILARITY_ZERO_CROSSING_RATE')
KeyError: 'zcr'
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/queues.py", line 268, in _feed
    send(obj)
IOError: [Errno 32] Broken pipe

bishoph, 2019-04-15:
Yes, one option is missing but there is already a pull request available:
https://github.com/bishoph/sopare/pull/55

a3020, 2019-04-15:
@bishoph Yes, I submitted that PR. But adding the config option with `0` doesn't solve this error.

bishoph, 2019-04-15:
Ah ok. Will fix this ASAP...stay tuned.

bishoph, 2019-04-15:
Hmm, with the config option the error does not occur. Make sure to use the config file either with the "-i" option or by changing the default.ini

a3020, 2019-04-15:
Is there caching going on somewhere? 

I'm checked out in `testing` now, and run `./sopare.py -p`. I don't get the error above. Then I comment the config setting and rerun. Still no error. Then I specify the config file with `-i config/default.ini` and still don't get the error. That's a mystery to me, because that config setting is now missing, so it *should* error out, right?

And just before I posted this, I ran it again (without changing anything) and now I do get the error again...

a3020, 2019-04-15:
In all cases I get the following error:

> INFO:sopare.recorder:start endless recording
INFO:sopare.processing:starting append mode
/usr/lib/python2.7/dist-packages/numpy/core/numeric.py:531: ComplexWarning: Casting complex values to real discards the imaginary part
  return array(a, dtype, copy=False, order=order)
INFO:sopare.processing:stop append mode because of silence

bishoph, 2019-04-15:
There is no caching and you can ignore the INFO messages and the warning. Will take a deeper look to fix the warning though.

bishoph, 2019-04-15:
Fixed warning with commit 9e5bff9 in testing branch.

closed 2019-04-15 by bishoph

a3020, 2019-04-15:
I tested with your commits, and I don't get an error now before the recording is stopped. However, right after it I still get:

> INFO:sopare.recorder:stop endless recording
Traceback (most recent call last):
File "/usr/lib/python2.7/multiprocessing/queues.py", line 268, in _feed
send(obj)
IOError: [Errno 32] Broken pipe

After the program exits. Does this ring a bell for you?

bishoph, 2019-04-15:
The different threads and the connecting queues are not in sync when shutting down. Commit b6eae99 should fix this. The used sleep is suboptimal but works for me.

a3020, 2019-04-15:
I pulled the changes from the `testing` branch, but the same error is still present. Do I need to do anything?

bishoph, 2019-04-17:
No. Are you using SOPARE with a Raspberry Pi and if so which version?

a3020, 2019-04-17:
Yes, Raspberry Pi 3 Model B+.

bishoph, 2019-04-19:
I could delay the queue close function even more but this is nothing I want to do. Will fix this eventually when I have a good idea...

a3020, 2019-04-20:
@bishoph Fyi, I set the sleep time to `10`, but it's still erroring out. (https://github.com/bishoph/sopare/commit/b6eae99e92b0dfe43a8e8a7b4c48687a59cae9ce#diff-32d135a4421f763f6c488918ce41046dR93)

bishoph, 2019-04-20:
Ok thx for the info. Are you using a custom config? If yes, please provide it or test with the default one...