bishoph/sopare #60: Is 'PROGRESSIVE_FACTOR' still used?

a3020, 2019-04-15:
I don't see the 'PROGRESSIVE_FACTOR' setting being referenced in the code. Am I missing something?

https://github.com/bishoph/sopare/blob/testing/config/default.ini#L40
https://github.com/bishoph/sopare/search?q=factor&unscoped_q=factor

bishoph, 2019-04-15:
Sure, it is in the code as a simple "grep" shows.

closed 2019-04-15 by bishoph

a3020, 2019-04-15:
@bishoph can you point me to the file + line number, please?

a3020, 2019-04-15:
Well, I don't see it. But again, maybe I'm missing something...

[image]

bishoph, 2019-04-15:
I don't see it either and I need a strong coffee ;) The variable PROGRESSIVE_FACTOR is not longer in use. Will remove it with one of the next commits.