bishoph/sopare #64: SOPARE PROCESSING PROBLEM

hankp46, 2019-04-17:
I am trying to get a GoPiGo Robot Car to respond to voice commands. I have some success with certain commands but on others it seems SOPARE does  not "listen (hear"). To wit:
I say 'light" get following:
('entering run', 'Wed Apr 17 10:15:39 2019')
('input', [u'light'])
leaving lights
('leaving run', 'Wed Apr 17 10:15:40 2019')
('entering run', 'Wed Apr 17 10:15:40 2019')
I say stop
('entering run', 'Wed Apr 17 10:15:49 2019')
('input', [u'stop'])
Robocar Stopping
Leaving Stop
('leaving run', 'Wed Apr 17 10:15:49 2019')
('entering run', 'Wed Apr 17 10:15:50 2019')
I say forward
('entering run', 'Wed Apr 17 10:16:10 2019')
('input', [u'forward'])
Robocar Moving Forward
Leaving Forward
('leaving run', 'Wed Apr 17 10:16:10 2019')
('entering run', 'Wed Apr 17 10:17:03 2019')
I say stop - nothing happens, car keeps going forward, continue saying stop,  here's printout
('entering run', 'Wed Apr 17 10:18:48 2019')
('entering run', 'Wed Apr 17 10:18:49 2019')
('entering run', 'Wed Apr 17 10:18:50 2019')
('entering run', 'Wed Apr 17 10:18:51 2019')
('entering run', 'Wed Apr 17 10:19:29 2019')
('entering run', 'Wed Apr 17 10:19:31 2019')
('entering run', 'Wed Apr 17 10:21:46 2019')
('entering run', 'Wed Apr 17 10:21:47 2019')
('entering run', 'Wed Apr 17 10:22:01 2019')
('entering run', 'Wed Apr 17 10:22:02 2019')
('entering run', 'Wed Apr 17 10:23:06 2019')
('entering run', 'Wed Apr 17 10:23:08 2019')

Only thing I see a difference in is that "light" and 'stop" are single actions while 'forward" is a continuous action (car keeps moving). I don' see why this would matter since in a program that takes keyboard input  it keeps accepting commands (forward and then stop0

Any advice would be greatly appreciated!

Here's the program:
#!/usr/bin/env python
# -*- coding: utf-8 -*-

"""
Copyright (C) 2015 - 2017 Martin Kauss (yo@bishoph.org)

Licensed under the Apache License, Version 2.0 (the "License"); you may
not use this file except in compliance with the License. You may obtain
a copy of the License at

 http://www.apache.org/licenses/LICENSE-2.0

Unless required by applicable law or agreed to in writing, software
distributed under the License is distributed on an "AS IS" BASIS, WITHOUT
WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied. See the
License for the specific language governing permissions and limitations
under the License.
"""
from gopigo import *
import sys
import time

def Forward():
    print("Robocar Moving Forward")
    fwd()
    print("Leaving Forward")

def Back():
    print("Robocar Moving backward")
    bwd()
    print("Leaving backward")

def Stop():
    print("Robocar Stopping")
    stop()
    print("Leaving Stop")

def Light():
    led_on(0)
    led_on(1)
    time.sleep(1)
    led_off(0)
    led_off(1)
    print("leaving lights")
    
def run(readable_results, data, rawbuf):
    print("entering run",time.asctime(time.localtime(time.time())))
    if (len(readable_results) > 2 or len(readable_results) == 0):
        return
    print("input",readable_results)
    try:
        if ('forward' in readable_results):
            Forward()
        elif ('back' in readable_results):
            Back()
        elif ('stop' in readable_results):
            Stop()
        elif ('light' in readable_results):
            Light()
    except Exception as err:
        print (err)
    print("leaving run",time.asctime(time.localtime(time.time())))    
    return

hankp46, 2019-04-17:
Could it be the noise of the motors drowning out the commands?

bishoph, 2019-04-17:
No idea as I have no further details about the environment, noise, used hardware, ... I may said this before: a good microphone, proper training in combination with a well set config could help.

To answer your question: with a rough mic and rough training a potential environmental sound could be interpreted as false positive.

hankp46, 2019-04-17:
Just to let you know, it was the noise from the motors, which are to close to the mic. Once i turned them off all commands worked properly. Could not set a threshold  that would not recognize noise but commands. Will need to look for a bluetooth mic that will be far from motors. Any suggestions on mic?

hankp46, 2019-04-22:
Is there any way of using input from a pipe instead of from a microphone?

closed 2019-04-29 by hankp46