bishoph/sopare #65: Strict length check not working correctly

a3020, 2019-04-19:
## Dictionary analysis

```
sonos uit
 minp [0, 0, 0, 0, 0, 0, 16, 0, 1, 9]
 maxcp [23, 14, 12, 15, 14, 3, 12, 6, 1, 3]
 peaks [[0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 15, 16], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13], [0, 1, 2], [16, 17, 18, 19, 20, 21, 22, 23, 36, 37, 39, 41], [0, 1, 3, 4, 5, 6, 7, 9, 12], [1], [9, 10, 11]]
 mincp [4, 1, 0, 0, 0, 0, 0, 0, 1, 3]
 min_tokens 6
 df [[27, 28, 29, 30, 31, 32, 37, 38, 49, 50], [22, 28, 29, 37, 39], [20, 28, 29, 30, 31, 426], [21, 22, 30, 31, 32, 273], [21, 22, 23, 31, 32, 217], [20, 21, 32, 217, 221, 224, 225, 237, 247], [112, 188, 217, 219, 224, 227, 237], [55, 64, 224], [64], [71]]
 max_tokens 10
 maxp [22, 16, 12, 15, 13, 2, 41, 12, 1, 11]
 cp [[15, 23, 17, 12, 14, 18, 4, 19], [11, 1, 8, 14, 7, 2, 12], [12, 0, 10, 8, 5, 11, 7], [9, 0, 11, 8, 15, 6, 7, 12], [10, 2, 0, 12, 14, 3, 13, 6], [0, 1, 3], [0, 12], [6, 0, 5], [1], [3]]
sonos aan
 minp [0, 0, 0, 0, 0, 0, 0, 1, 0]
 maxcp [17, 14, 12, 13, 14, 6, 0, 22, 0]
 peaks [[0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 26], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13], [0, 1, 2, 3, 4, 5, 6, 8], [], [1, 5, 6, 7, 8, 9, 10, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 28, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 46], []]
 mincp [2, 5, 3, 1, 2, 0, 0, 0, 0]
 min_tokens 6
 df [[25, 28, 29, 30, 31, 33, 53, 56, 58], [26, 27, 28, 29, 30, 31, 32, 54], [21, 28, 29, 31], [20, 21, 23, 26, 28, 29], [21, 26, 28, 30, 31, 32, 64], [20, 22, 30, 31, 32, 48], [218, 219, 224, 234, 236, 250, 407], [56, 104, 106, 124, 218, 234], [218]]
 max_tokens 9
 maxp [26, 14, 12, 12, 13, 8, 0, 46, 0]
 cp [[15, 11, 2, 4, 8, 7, 17], [14, 10, 8, 12, 11, 5, 7], [12, 10, 3, 11, 6, 8], [6, 4, 11, 3, 9, 10, 13, 1], [2, 5, 12, 6, 11, 14, 13, 3], [6, 1, 0, 3], [0], [18, 0, 9, 7, 22], [0]]
```

As you can see, I have two entries here. Both have a minimum of 6 tokens.

## Listening mode

```
framing_match: [[[0.9349708657562561, 0.9312877744953045, 0.6133550110770433, 0, 8, u'sonos uit'], [0.90176585052958469, 0.7789798823771097, 0.5701230965830532, 0, 6, u'sonos uit'], [0.94950583706564162, 0.9154548446469193, 0.6428323066535573, 0, 7, u'sonos uit'], [0.95076525041769766, 0.8795897527611977, 0.6051427383275321, 0, 7, u'sonos uit'], [0.95007502821741097, 0.8387421369666258, 0.6556607501663103, 0, 7, u'sonos uit'], [0.95827391263989214, 1.0368974447722263, 0.7402093205655975, 0, 8, u'sonos uit'], [0.91780630605262525, 0.7510179212571506, 0.6369058011814274, 0, 6, u'sonos uit'], [0.95674429568251573, 0.8624356787606071, 0.6783741092700513, 0, 7, u'sonos uit'], [0.93430490017842738, 0.7920414529466285, 0.5847175192395458, 0, 9, u'sonos uit'], [0.95413022451045415, 0.927029169917757, 0.6713787114792085, 0, 7, u'sonos uit']], [[0.96598702231836697, 0.6673304237567618, 0.5011630286500125, 0, 6, u'sonos aan'], [0.92453022364974258, 0.8041217751739936, 0.5830860568091515, 0, 6, u'sonos aan'], [0.95497585654218953, 0.8970571073694553, 0.7358320891453005, 0, 8, u'sonos aan'], [0.96564294626270653, 0.9519829951416238, 0.6924445622253961, 0, 8, u'sonos aan'], [0.937519246856095, 0.8742098647124661, 0.5574575452183169, 0, 8, u'sonos aan'], [0.96593414244064724, 0.9652981801038191, 0.6249335253882848, 0, 8, u'sonos aan'], [0.95577579961969994, 0.984155422390141, 0.7440879290128566, 0, 8, u'sonos aan'], [0.94298081805181178, 0.9562129722054019, 0.5966369560839504, 0, 9, u'sonos aan'], [0.923157366406122, 0.6279331276063901, 0.5331500190913231, 0, 6, u'sonos aan'], [0.94889193553841644, 0.9131650587619591, 0.7565006056552142, 0, 8, u'sonos aan']]]

sorted_best_match: [[0.923157366406122, 0.6279331276063901, 0.5331500190913231, 0, 6, u'sonos aan'], [0.96598702231836697, 0.6673304237567618, 0.5011630286500125, 0, 6, u'sonos aan'], [0.90176585052958469, 0.7789798823771097, 0.5701230965830532, 0, 6, u'sonos uit'], [0.93430490017842738, 0.7920414529466285, 0.5847175192395458, 0, 9, u'sonos uit']]

match_results: [u'sonos aan', u'sonos aan', u'sonos aan', u'sonos aan', u'sonos aan', u'sonos aan', u'sonos uit', u'sonos uit', u'sonos uit']

STRICT_LENGTH_CHECK failed for sonos uit: 6 > 3 < 10
```

Based on the above, I conclude that only one of the two words is excluded as a match. If I read the debug mode correctly, the input is "3" tokens. The input I used was the word "aan". 

In my opinion, **both match results should be excluded.**
Can you confirm this, or am I misinterpreting the debug messages?

bishoph, 2019-04-19:
As you can see in the match_results for "sonos aan" is "6" and "3" for "uit". That's why the strict length check is sorting out "uit". You can't compare dict entries and live sound analysis as it is like comparing an apple and a painting of an apple.

closed 2019-04-19 by bishoph

a3020, 2019-04-19:
I only have two entries; "sonos aan" and "sonos uit". I don't have an entry "uit". I'm afraid I don't understand what you mean. It's odd to me that if I use "aan" as input 'sound', it's triggering the dictionary entry "sonos aan", even though that phrase is way longer. Is there something obvious I'm missing?

For reference:
```
self.debug_info += 'STRICT_LENGTH_CHECK failed for '+result[0] + ': ' + str(self.dict_analysis[result[0]]['min_tokens']) + ' > ' + str(len(result)) + ' < ' + str(self.dict_analysis[result[0]]['max_tokens']) + '\n'
```

bishoph, 2019-04-19:
You should simply replace "uit" with "sonos uit" in my last comment ;)

Ok, as you are struggling you may want to train the 3 words separately:

1) sonos
2) uit
3) aan

Here is my advice: make sure you get decent results for each of the words. Try to adjust your config and the settings, then train and  test. Document your config and your results. Then delete all trained stuff and repeat until you get the best results. You have to explore the best settings as it's your hardware and your environment. If you can't get good results you may want to try different microphones and start again with the default config...

a3020, 2019-04-20:
> make sure you get decent results

I trained the words separately. Now it triggers 'sonos aan' when I say 'uit'. Haha! I think the verbose mode could be improved with some guidance. It's hard for me to decide when a result is 'decent'. I'm using a sing microphone from very short distance, so I don't think background noise is a problem. When I record in Audacity, the waves look really good actually.

bishoph, 2019-04-20:
Nice looking waves are good but adjusting the config, train, test and repeat does in most cases the trick. First find the best THRESHOLD. Next find your optimal timing values. Create a wav file from the training and test if it contains the right stuff. Use the write/read options of SOPARE in combo with the plot option  to see a graphical interpretation of your input. Then you may want to test around the characteristic options and finally you get to the compare section. A couple of blog posts and videos with further instructions are available.

a3020, 2019-04-25:
> Create a wav file from the training and test if it contains the right stuff.

How can you make a .wav file out of a .raw file?

bishoph, 2019-04-25:
Like described in the help and/or the git "Usage" section: With "-~" or "--wave"...