bishoph/sopare #71: Sopare with Pipe Input

hankp46, 2019-05-13:
I'm trying to run sopare using a "virtual microphone" for input. This is basically a pipe. I know the input contains the word "light" (created a .wav file from the pipe).
When I run 
```
pi@raspberrypi:~/robocar/sopare $ ./sopare.py -l
sopare 1.5.0
```
These are the results
framing_match: [[[0.90861854996172831, 1.391658788483111, 1.0512859126694987, 9, 9, u'light'], [0.9122890274553479, 1.4619765636975264, 1.1070853044823012, 9, 9, u'light'], [0.87645107084278162, 1.5528330304814562, 1.3575703756880773, 9, 6, u'light'], [0.87042081118285985, 1.4661159524676999, 1.2735323762277182, 9, 8, u'light'], [0.89475312080401992, 1.5521309144655477, 1.0314636350967308, 9, 9, u'light'], [0.85302260879552649, 1.4150844813150159, 1.0970118806214015, 9, 5, u'light'], [0.89072533446402702, 1.387208040005574, 1.0938840145117203, 9, 9, u'light'], [0.86778785963393557, 1.5621954822883861, 1.165076978205376, 9, 6, u'light'], [0.8687923928763821, 1.480054708239093, 1.0340545317795842, 9, 6, u'light'], [0.88771104442424231, 1.3565261436458582, 1.06713360465813, 9, 9, u'light'], [0.8320172238950867, 1.6677398553347558, 1.2535348213012358, 9, 5, u'light'], [0.88970577279893903, 1.5046103544343876, 1.3096829321917247, 9, 9, u'light'], [0.88726174667037316, 1.4688747060807568, 1.2127978890925715, 9, 6, u'light'], [0.88079012404318247, 1.3225640272985575, 1.2143986991195164, 9, 5, u'light'], [0.89559257261082159, 1.390214528060448, 1.1766441271805355, 9, 8, u'light'], [0.91967328346690935, 1.4046702543710126, 1.1751282387702247, 9, 6, u'light'], [0.90971114786550544, 1.4255651499315243, 1.2460328765744622, 9, 9, u'light']], [[0.88873421928104379, 1.3210478318403602, 1.0135655862000394, 8, 9, u'light'], [0.91198079923076403, 1.356571946518401, 1.0274923198693091, 8, 10, u'light'], [0.90133878987907712, 1.4818260417771907, 1.3078490344587563, 8, 6, u'light'], [0.87694701915129791, 1.4222890626925506, 1.2861547844761032, 8, 8, u'light'], [0.87477993876322468, 1.5806248953875217, 1.0425717032325406, 8, 9, u'light'], [0.87394162835733324, 1.3907085871912472, 1.1582702418322486, 8, 5, u'light'], [0.89166504125446067, 1.3216742955642895, 1.0485184962927963, 8, 9, u'light'], [0.83378964916143783, 1.5289474815365647, 1.1351921021700466, 8, 6, u'light'], [0.85340527428499569, 1.4659174601415714, 1.0398483495491306, 8, 6, u'light'], [0.90534376143482953, 1.2616749458736802, 1.0184403654742076, 8, 9, u'light'], [0.85032808877152155, 1.5239744234947927, 1.1855192540490342, 8, 5, u'light'], [0.89846088413794289, 1.4773183456453935, 1.3111490730252067, 8, 9, u'light'], [0.8936120642795321, 1.3642031020354375, 1.161616582429893, 8, 6, u'light'], [0.88679762827639441, 1.3052485965758147, 1.2854810667875416, 8, 5, u'light'], [0.9132258639663291, 1.3341074056637183, 1.169330169578435, 8, 8, u'light'], [0.90771804019917168, 1.3865896186576128, 1.251722587436895, 8, 6, u'light'], [0.91362409655999088, 1.381889443879187, 1.178421470472468, 8, 9, u'light']]]

sorted_best_match: [[0.90534376143482953, 1.2616749458736802, 1.0184403654742076, 8, 9, u'light'], [0.88873421928104379, 1.3210478318403602, 1.0135655862000394, 8, 9, u'light'], [0.88771104442424231, 1.3565261436458582, 1.06713360465813, 9, 9, u'light'], [0.90861854996172831, 1.391658788483111, 1.0512859126694987, 9, 9, u'light']]

match_results: ['', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '']
```
I then attempted the following

```
pi@raspberrypi:~/robocar/sopare $ ./sopare.py -r /home/pi/testout.raw
sopare 1.5.0
framing_match: [[[0.93143922824859005, 1.6296656295969674, 0.8337357094073549, 10, 7, u'light'], [0.93809562163564031, 1.4609089255910317, 0.7973416468100594, 10, 7, u'light'], [0.94870481611418012, 1.3790384629210912, 0.9107312352603915, 10, 6, u'light'], [0.90518180809783577, 1.6090526190409575, 1.1515758986843097, 10, 7, u'light'], [0.93506509543667815, 1.5105002405038637, 0.8968463213467847, 10, 7, u'light'], [0.93909019415410422, 1.338981211043447, 1.0796311850181273, 10, 5, u'light'], [0.91546848533198433, 1.562262316445043, 0.9252254382874258, 10, 7, u'light'], [0.91458321003542142, 1.4684301295123108, 0.8305982746637485, 10, 6, u'light'], [0.92373830135766377, 1.410970968305797, 0.8349829688346478, 10, 6, u'light'], [0.92496723450302298, 1.6292112121407563, 1.0419468151252544, 10, 7, u'light'], [0.92961619203351042, 1.549213826962881, 0.9372459289602949, 10, 5, u'light'], [0.9205794968808344, 1.5721044316673825, 1.0182074907115457, 10, 7, u'light'], [0.94167414208275402, 1.370267971234086, 0.9608187060608788, 10, 6, u'light'], [0.94734343643855345, 1.4241087850171124, 1.1858712395297095, 10, 5, u'light'], [0.93777880708384143, 1.5053770959186088, 1.099951604576774, 10, 7, u'light'], [0.95481024459834529, 1.3232937823398832, 0.952445162676284, 10, 6, u'light'], [0.93247494403374354, 1.4943157594733714, 0.9412592544895736, 10, 7, u'light']], [[0.93871034988318891, 1.5462147150784542, 0.8670528775888866, 9, 8, u'light'], [0.92892115076121451, 1.43112415095734, 0.8682052856071316, 9, 8, u'light'], [0.95951226306262949, 1.3501279821204972, 1.0217430642635221, 9, 6, u'light'], [0.93320665595497609, 1.5432106321830708, 1.33068131970698, 9, 8, u'light'], [0.92938501006142948, 1.4827450929273778, 1.0197505473132835, 9, 8, u'light'], [0.95583723052071645, 1.3093336721616384, 1.034083357991729, 9, 5, u'light'], [0.92889203058023551, 1.5384109437880351, 1.0185593958530414, 9, 8, u'light'], [0.91818975238487088, 1.4361373132851234, 0.822119315845041, 9, 6, u'light'], [0.92327823927011499, 1.363507757160412, 0.7781133264190919, 9, 6, u'light'], [0.92542807530632143, 1.5283358841514343, 1.0797419294358719, 9, 8, u'light'], [0.95076901953309712, 1.5197819961380312, 0.9863621998224366, 9, 5, u'light'], [0.92960276938274344, 1.535534091944387, 1.1144580120381622, 9, 8, u'light'], [0.93567318173345815, 1.3508502148196515, 1.0047276452675853, 9, 6, u'light'], [0.9546711119028215, 1.4011303010608496, 1.0978964684098096, 9, 5, u'light'], [0.94371555961522835, 1.4456610209825784, 1.1938785062268364, 9, 8, u'light'], [0.95605044440879705, 1.2768031990934254, 0.979769102951308, 9, 6, u'light'], [0.9423548919772553, 1.4841402665262804, 1.1399464325991953, 9, 8, u'light']]]

sorted_best_match: [[0.92327823927011499, 1.363507757160412, 0.7781133264190919, 9, 6, u'light'], [0.92373830135766377, 1.410970968305797, 0.8349829688346478, 10, 6, u'light'], [0.95605044440879705, 1.2768031990934254, 0.979769102951308, 9, 6, u'light'], [0.93809562163564031, 1.4609089255910317, 0.7973416468100594, 10, 7, u'light']]

match_results: ['', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '']
```
It seems that in both case the pipe (or .raw file) is being read but sopare cannot find the match.
Looking for any ideas.

hankp46, 2019-05-13:
Could it be that my dictionary  entries came from true microphone input and I am comparing to computer generated audio?

bishoph, 2019-05-13:
No idea, really. Best way to approach is in my IMHO to do a full re-training as described in the usage section. Delete your training files and the dictionary entries with the following commands:

./sopare.py -d "*"
rm dict/*.raw

If you changed config options you may want to consider to start with the default values.

After that, train, build your dict and test again.

hankp46, 2019-05-13:
Once I trained it from the raw data and then used the raw data as input to sopare it worked perfectly. Thanks!

closed 2019-05-13 by hankp46