bishoph/sopare #72: Run For A Single Event

hankp46, 2019-05-15:
Is there a way to run Sopare so it stops after recognizing a single word?
Alternatively is there a way to put a time limit on the -l argument?

bishoph, 2019-05-15:
If you are using a script you could terminate the python processes the script started. Maybe in combination with a time or an desired output. The alternative is to use your custom plugin code and use sys.exit() ...

hankp46, 2019-05-16:
This is the command I'm running 
./sopare.py -l  -r testout.raw

I get the proper  results  but the process hangs. I tried putting sys.exit () in my plugin code. But it still hangs.
ThenI I tried putting sys.exit() in various logical locations in sopare.py  and recorder.py with same results. It seems that the combination of looping  and reading  a file won't allow the  sys.exit to exit all process , or there is another process that's causing the hang up ( just a guess).

Any ideas?

BTW - I can kill it in a script with a timer but this makes the rest of my project more complicated,

Thanks for all the great work you have done!

bishoph, 2019-05-16:
The loop option and the read option are not combinable. After reading a file SOPARE stops automatically. The loop option is, like the documentation says, endless. The issue with the "hanging" is to the fact that SOPARE uses different independent processes and the process that should be terminated first is the read process and not the analyzing/plugin process. Saying this I have to revoke my suggestion to put the exit code into your plugin.

hankp46, 2019-05-26:
Solved

closed 2019-05-26 by hankp46