bishoph/sopare #77: Questions regarding the use of Sopare

shandrios, 2019-07-09:
Hello again, still a big fan. I've been tinkering around a bit with your program, and I've got some questions.

1. By default, the frequency range used in the speech analysis is 20-600, even though human speech goes much higher than that. Does increasing this range have an effect on the accuracy of the analysis? Also, it seems like the built-in FFT plot function's x axis caps out at 2000Hz by default, even if the frequency range is e.g 20-5000. Does the actual analysis still take the entire range into account, and can the axis limits be increased?

2. I noticed the master branch hasn't had a commit since January 2018. Have there been any significant improvements in the testing branch that would warrant using that over the master branch for my own project?

3. In my own project, I would like my Raspberry to recognise simple voice commands from a handful of different people. I can't necessarily get word samples from all of them, so I'm wondering what settings in the config file I should tinker with in order to improve the rate of success.

Thanks in advance, have a good day.

bishoph, 2019-07-09:
1) Yes, more frequencies means normally more precision. But the default range gives decent results already even with normal hardware and works also for some range. 
The plot scales automatically and shows the time domain as well as the frequency domain.

2) Testing branch is ahead of master. Mostly smaller bugfixes. You get the full view right here: https://github.com/bishoph/sopare/compare/testing
Switching branches is easy so you can give it a try.

3) Test. Adapt. Repeat. Can't really give better advice without details ;)

shandrios, 2019-07-10:
My issue is that the plots don't seem to scale past 2000Hz. In the attached image, you can see that I've set the HIGH_FREQ and START_PROGRESSIVE_FACTOR to 5000, but the plots still only shows up to 2000/400. Is there something else I need to do to get the full range?
[image]

bishoph, 2019-07-10:
Check if your hardware limits the input. Other than that it could be that you are using a different configuration while you are using plot. Or it is a bug ;).

shandrios, 2019-07-11:
Probably a bug then, because that is the only config file I have, and the Full FFT graph shows frequencies all the way up to 20000.
[image]

On another note, is it possible to use the volume of the analyzed sound in  a plugin? I noticed that the plugin's run function takes three parameters, but only readable_results is used in your examples. Can the volume of the sound be gotten from the rawbuf or data parameters?

kimgenegaby, 2019-12-08:
I noticed you concentrated lots of time in YouTube in the frequency domain. How to get frequency domain with a wav file?

dumblob, 2019-12-28:
The question about frequency range would interest me as well. Any insights?

bishoph, 2019-12-28:
What question about frequency range is unanswered?

dumblob, 2019-12-28:
Question about how is the following possible:

>Probably a bug then, because that is the only config file I have, and the Full FFT graph shows frequencies all the way up to 20000.

(see 3 comments above in https://github.com/bishoph/sopare/issues/77#issuecomment-510472815 )

bishoph, 2019-12-28:
Only the full FFT graph shows all frequencies. Single token graphs are, as the word states, tokenized and inherit only parts of the frequencies. Like a single piece of cake don't contain all the ingredients of the full cake...

dumblob, 2019-12-28:
>Single token graphs are, as the word states, tokenized and inherit only parts of the frequencies.

That didn't appear to me. Thanks for clarification and a great project overall!