bishoph/sopare #79: recognition gone rogue

classicryder, 2019-11-01:
I notice that after some time, the sound pattern recognition can go rogue, in more details i mean it randomly picks up any sound and recognize it as a trained word, even though there is no sound/noise at all. This behavior is occurring randomly (not every day) but mostly during midnight, i have no idea what happened, usually when it happened, i manually stopped sopare, but the very strange thing is, eventhough i restart the raspberry pi, it still occurs. When i shutdown the pi, and start it up again in the morning, it behaves properly again

i thought it was a microphone problem but im not sure, because it happens randomly with no pattern

any advice?

bishoph, 2019-11-01:
If it happens frequently and even in a fixed time span I would place my bet on external circumstances like windy conditions or alike. I had the same in some spots where the mic was under the roof and the wind created patterns in the low bandwidth which triggered or hindered the recognition. In my case the solution was to increase the THRESHOLD a bit. This helped in terms of false positives but the environmental noise still holds up the recognition sometimes.

This real-time audio visualization tool helped my to figure it out:
https://soundlab.cs.princeton.edu/software/sndpeek/

classicryder, 2019-11-01:
r u running this soundlab software from the pi?

bishoph, 2019-11-01:
No, it's running on a laptop ;)