bishoph/sopare #80: Beginner Guide missing

SassNinja, 2019-11-06:
Hi,

I haven't much experience with raspberry pi yet.
Though I wanna have some simple voice control, running 24/7, to call an API on specific words.

What I've got so far:
- python code that makes the desired API requests
- training files for the desired words (working more or less reliable)

For testing I enter in terminal (while connected via ssh) this
`./sopare.py -l`
and then say `licht an` what gets recognized correctly and reults in the desired action

But of course I want it to automatically listen without being forced to always start the script myself.
Since your README says
> Runs on several PIs 24/7 and controls smart home devices

I was hoping to find some information regarding how to make the raspberry always listen to my voice - but I haven't been lucky (neither on github nor your website)

Would you explain me how to do this?
Some example would help me a lot.

Apart from that I've noticed the python script freezes after a while what's of course not suitable for 24/7 usage.
Is there anything else I need to consider (in case the question above doesn't answer this)

bishoph, 2019-11-07:
There is lots of stuff on https://bishoph.org about SOPARE and how to use it even if you have never used it. But, you need some experience and knowledge as you maybe want to copy/adapt my existing init.d script (for 24/7) or to write custom code (start desired action). 

Here is the link to the init.d script:
https://github.com/bishoph/Misc/blob/master/sopare_service

SassNinja, 2019-11-08:
Hi @bishoph 

thanks for the init.d script!

What I tried to so far is adding the following to `/etc/rc.local`
```
cd /home/pi/dev/sopare
./sopare.py -l &
```

Easy as pie and working more or less: the script starts now on reboot.
Though your init.d script looks more... professional and suitable for production.
I'll try to dig into it.

closed 2019-11-08 by SassNinja

bishoph, 2019-11-08:
The point of the init.d script is that you can start/stop  SOPARE from anywhere with the following commands:

sudo service sopare_service start
sudo service sopare_service stop

It is even possible to request the status:

sudo service sopare_service status

Quick read through: https://www.geeksforgeeks.org/run-levels-linux/

Have fun!