bishoph/sopare #82: How to properly delete a word from the dictionary?

ceyhunemreozturk, 2019-12-08:
Hi,

In the article with the link https://www.bishoph.org/step-by-step-raspberry-pi-offline-voice-recognition-with-sopare/, it is stated that to the following commands must be used to delete all training files:

rm dict/*.raw
./sopare.py -d "*"

By inspecting these commands, I thought that the following steps would be enough to remove a word from the dictionary and continue using the dictionary with other remaining words:

Step 1: Use the following command to find the names of the raw files of a word (For example, "test"):
grep -rnwl "dict" -e "test"

Step 2: Remove these files with rm command. 

Step: 3: ./sopare.py -d "test"

Step 4: ./sopare.py -c 

These steps seemed to work in the case where I deleted a word which was recorded after the latest use of the following command: ./sopare.py -c (It is a word which was not compiled.). However, after I deleted a word which was compiled and got listed in the dict.json file, I got the following error whenever I used the command "./sopare.py -c ":

sopare 1.5.0
recreating dictionary from raw input files...
Traceback (most recent call last):
  File "./sopare.py", line 187, in <module>
    main(sys.argv[1:])
  File "./sopare.py", line 100, in main
    recreate_dict(debug, cfg)
  File "./sopare.py", line 130, in recreate_dict
    utilities.recreate_dict_from_raw_files()
  File "/home/pi/Desktop/sopare/sopare/util.py", line 195, in recreate_dict_from_raw_files
    self.writeDICT(self.getCompiledDict())
  File "/home/pi/Desktop/sopare/sopare/util.py", line 156, in getCompiledDict
    json_obj = json.load(raw_json_file, object_hook=sopare.numpyjsonencoder.numpyjsonhook)
  File "/usr/lib/python2.7/json/__init__.py", line 291, in load
    **kw)
  File "/usr/lib/python2.7/json/__init__.py", line 352, in loads
    return cls(encoding=encoding, **kw).decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 380, in raw_decode
    obj, end = self.scan_once(s, idx)
ValueError: Expecting object: line 1 column 33063 (char 33062)

I tried deleting the dict.json file and using the dict.json file that is posted on GitHub. However, I am still getting the same error. Can you share with us how to properly delete a word from the dictionary?

Thanks in advance.

bishoph, 2019-12-09:
You need to delete the raw file(s). Either delete all raw files by using "rm dict/*.raw" or find the corresponding file by using the "overview" option (./sopare.py -o) and delete single files with the second token which identifies the raw file.

ceyhunemreozturk, 2019-12-09:
Thanks for the help, sir. I hope you keep doing great works. Before closing the issue, I'd like to ask one more thing about the error that I got here. I got this error for following the steps that I mentioned here. Is there any way of moving the raw files of my ruined project to a new SOPARE project and use this training data in the new project? I tried to use my raw files on a fresh clone of the project, however I was not able to compile the dictionary. The answer to this question maybe also useful for using the same training data in multiple devices.

bishoph, 2019-12-09:
With the same hardware/setup you should be able to use and move the raw files generated by SOPARE, but using different hardware/setup/environment I would start from scratch.

ceyhunemreozturk, 2019-12-09:
I now found out that the reason why I was not able compile the dictionary was the fact that the threshold parameter in the two projects were different. After changing the default.ini file, I was able to use the old training files. Thanks for the help!

closed 2019-12-09 by ceyhunemreozturk