bishoph/sopare #83: Wrong spoken word.

kimgenegaby, 2019-12-09:
I noticed you concentrated lots of time in YouTube in the frequency domain. How to get frequency domain with a wav file? Never mind. I compiled the Fscape Java program from Github( by three way, this contributor is also from German. Right?) to have frequency domain.
    Another question I have is that "sorted_best_match" was 98%, but the "match_result" shows {u'test'} not the right spoken word as you had in the YouTube.

bishoph, 2019-12-09:
Not sure what happened but maybe the words are very similar and the best prediction was used and displayed...

kimgenegaby, 2019-12-11:
Hello Bishoph,

     Thanks Bishoph for the fast response. I am very grateful for you to give out the free Github python programs. Is there a way to give a donation?
     It will be wonderful if you can provide another video for beginners from the beginning to the end. After you showed the video referring to frequency transform of the microphone, I might have a problem with my microphone after I frequency transformed my recorded message. Therefore, I will buy a microphone that has better recording quality knowing I don't have to rely on online Google's speech to text. 
  
Again thank you.

Steve

bishoph, 2019-12-11:
Glad you find my work useful. 
There is currently no way for donations - but I'll think about it :)

Have fun!

kimgenegaby, 2019-12-11:
Hello Bishoph,

     When you are ready for a donation,
just email me!

Have a good day!

Steve