bishoph/sopare #85: Fabian Weller Fehlermeldung und Config-Datei Sopare

Fabian2305, 2020-03-31:
######################################################
# Stream prep and silence configuration options #########
#########################################################

[stream]

# Read chunk size
CHUNK = 512

# Sample rate
SAMPLE_RATE = 48000

# Volume threshold when audio processing starts / silence
THRESHOLD = 480

# Silence time in seconds when analysis is called
MAX_SILENCE_AFTER_START = 1.4

# Time in seconds after the analysis is forced
MAX_TIME = 2.4

# Start the analysis after reaching LONG_SILENCE
LONG_SILENCE = 20

# Characteristic length
CHUNKS = 3072

#########################################################
# Characteristic configuration options ##################
#########################################################

[characteristic]

# Steps boil down the data into smaller chunks of data.
# Smaller steps mean more precision but require
# normally more learned entries in the dictionary.
# Progressive value is used if you want to pack not
# so relevant frequencies
PROGRESSIVE_FACTOR = 0
START_PROGRESSIVE_FACTOR = 600
MIN_PROGRESSIVE_STEP = 5
MAX_PROGRESSIVE_STEP = 5

# Specifies freq ranges that are kept for further
# analysis. Freq outside of the ranges are set to zero.
# Human language can be found between 20 and 5000.
LOW_FREQ = 20
HIGH_FREQ = 600

# Make use of Hann window function
HANNING = true

# Range factor for peaks
PEAK_FACTOR = 0.7

#########################################################
# Compare configuration options #########################
#########################################################

[compare]

# Min. number of tokens to identify the beginning of a word
MIN_START_TOKENS = 3

# Min. value for potential beginning of a word
MARGINAL_VALUE = 0.7

# Minimal similarity across all comparison to
# identify a complete word across all tokens
MIN_CROSS_SIMILARITY = 0.8

# Calculation basis or token/word comparison
SIMILARITY_NORM = 0.6
SIMILARITY_HEIGHT = 0.4
SIMILARITY_DOMINANT_FREQUENCY = 0

# Number of best matches to consider.
# Value must be > 0
# If not specified or value < 1 value is set to 1
NUMBER_OF_BEST_MATCHES = 2

# Min. distance to keep a word
MIN_LEFT_DISTANCE = 0.9
MIN_RIGHT_DISTANCE = 0.7

# Use given number as results to assembly result
# 0 for all predictions
MAX_WORD_START_RESULTS = 2
MAX_TOP_RESULTS = 3

# Enable or disable strict length check for words
STRICT_LENGTH_CHECK = true
# Value to soften the strict length check a bit to still
# get quite precise results but to be less strict
STRICT_LENGTH_UNDERMINING = 2

# Short term memory retention time in seconds. Zero to disable STM
STM_RETENTION = 1.2

# Fill result percentage
# 0.5 means that half of the values can by empty to still get valid results
# A lower value should theoretically avoid false positives
FILL_RESULT_PERCENTAGE = 0.1

#########################################################
# Misc configuration options ############################
#########################################################

[misc]

# Loglevel (CRITICAL, ERROR, WARNING, INFO, DEBUG)
LOGLEVEL = DEBUG

#########################################################
# Experimental configuration options ####################
#########################################################

[experimental]

# Additional FFT analysis and comparison for CHUNKS/2 length
FFT_SHIFT = false

**Meine aktuelle Fehlermeldung:**

Traceback (most recent call last):
  File "/home/pi/dev/sopare/sopare/recorder.py", line 44, in __init__
    self.recording()
  File "/home/pi/dev/sopare/sopare/recorder.py", line 88, in recording
    buf = self.stream.read(self.cfg.getintoption('stream', 'CHUNK'))
  File "/home/pi/dev/sopare/sopare/config.py", line 36, in getintoption
    return self.config.getint(section, option)
  File "/usr/lib/python2.7/ConfigParser.py", line 359, in getint
    return self._get(section, int, option)
  File "/usr/lib/python2.7/ConfigParser.py", line 356, in _get
    return conv(self.get(section, option))
  File "/usr/lib/python2.7/ConfigParser.py", line 616, in get
    value = d[option]
  File "/usr/lib/python2.7/ConfigParser.py", line 573, in __getitem__
    return mapping[key]
MemoryError

bishoph, 2020-03-31:
Please attach the config file instead of posting in editor.

Thx.

bishoph, 2020-03-31:
Also, what operating system / version are you using. Run 
<code>
free
</code>
and post the output.
In addition, start SOPARE and run the following command while SOPARE is running and the error shows up and post the output:
<code>
ps auxf | grep sopare
</code>