bishoph/sopare #87: regarding pi4 and python3

umsaenga, 2020-07-30:
`Python3 compatibility and testing, install guides`

Wanted to inform you I was able to successfully install and run on a pi 4 model B and raspian OS lite 
Version: May 2020
Release date: 2020-05-27
Kernel version: 4.19

from the current instructions.

bishoph, 2020-07-30:
Cool. Thx for the info!

bishoph, 2020-07-30:
But Python3 still requires some work (hope I can work on this topic this fall), JFYI.

umsaenga, 2020-07-30:
So far it's working fine for me, i can train it and now testing commands on a breadboard.  No issues so far

umsaenga, 2020-07-30:
excellent work!! thank you for it. btw

bishoph, 2020-07-30:
Glad that you like it and you are welcome :)

tzukpolinsky, 2020-09-03:
HI
where does the Python3 compatibility and testing, install guides is?
i would like to try it myself as well

umsaenga, 2020-09-22:
> 
> 
> HI
> where does the Python3 compatibility and testing, install guides is?
> i would like to try it myself as well

I simply followed the instructions in the current readme file and replaced the appropriate python3 commands.

bishoph, 2021-01-01:
The current testing branch should work with Python 2 and 3, SOPARE version >= 1.5.6. In addition the readme contains the requirements for both versions.

closed 2021-01-01 by bishoph