bishoph/sopare #91: output is showing UNICODE, not the given input

nahianushan, 2021-03-08:
it don’t recognize Bangla word.
Whenever i train it named "১" [ "১“ means one ]
it shows "\u09e7“ instead of "১“
and also "\u09e8" instead of "২"

[image]

Can you help me to get rid of this situation, thank you

bishoph, 2021-03-08:
The output in your screenshot shows a debug message  A debug message is meant to help to debug the code and find/understand issues. In this special case the array contains a string encoded in Unicode. The array is printed in it's raw form and this is because the developer wants to see the original, unconverted content of the array. This is a feature and not a bug. If you want to print the output decoded you can achieve this by writing your own plugin.

bishoph, 2021-03-08:
The output comes from the default print plugin (https://github.com/bishoph/sopare/blob/testing/plugins/print/__init__.py) which provides the simplest form of an example. Feel free to change/enhance/optimize the default plugin and/or write your own custom plugin to print the content of the array in your favorite form.