bishoph/sopare #92: sopare in python3

yuzl340822, 2021-03-29:
Hello! I'm a student. I just started to try sopare on a raspberry pi. But I need to use Python 3 . I want to ask if there are some convenient ways to make the project run in Python 3. To be honest, I'm not very proficient in Python yet. Could you list some items? In addition, I heard that version 1.5.6 for Python 3 is already in the plan. I'd like to ask about the progress so far.Any way, Thank you and your project for helping me a lot.

bishoph, 2021-03-29:
Hi, the latest test branch is Python 3 compatible and should work:

https://github.com/bishoph/sopare/tree/testing

It also contains the updated requirements and steps to install...

Have fun!

yuzl340822, 2021-03-29:
Wow! Thank you again for your help！