bishoph/sopare #93: A Error about JSONEncode

yuzl340822, 2021-03-31:
I met a Error when running sopare/jump y is on encoder. py.  This is the code and the Error. 
`
class numpyjsonencoder(json.JSONEncoder):

    def default(self, obj):
        if isinstance(obj, numpy.ndarray):
            if obj.flags['C_CONTIGUOUS']:
                obj_data = obj.data
            else:
                cont_obj = numpy.ascontiguousarray(obj)
                if (cont_obj.flags['C_CONTIGUOUS']):
                    obj_data = cont_obj.data
                else:
                    raise Exception("numpyjsonencoder err: C_CONTIGUOUS not present in object!")
            data_base64 = base64.b64encode(obj_data)
            return dict(__ndarray__= data_base64, dtype = str(obj.dtype), shape = obj.shape)
        return json.JSONEncoder.default(self, obj)`

`Traceback (most recent call last):  File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap    self.run()  File "/home/pi/dev/sopare/sopare/worker.py", line 125, in run    self.util.store_raw_dict_entry(self.cfg.getoption('cmdlopt', 'dict'), self.raw_character)  File "/home/pi/dev/sopare/sopare/util.py", line 108, in store_raw_dict_entry    json.dump(json_obj, json_file, cls=sopare.numpyjsonencoder.numpyjsonencoder)  File "/usr/lib/python3.7/json/__init__.py", line 179, in dump    for chunk in iterable:  File "/usr/lib/python3.7/json/encoder.py", line 431, in _iterencode    yield from _iterencode_dict(o, _current_indent_level)  File "/usr/lib/python3.7/json/encoder.py", line 405, in _iterencode_dict    yield from chunks  File "/usr/lib/python3.7/json/encoder.py", line 325, in _iterencode_list    yield from chunks  File "/usr/lib/python3.7/json/encoder.py", line 405, in _iterencode_dict    yield from chunks  File "/usr/lib/python3.7/json/encoder.py", line 439, in _iterencode    yield from _iterencode(o, _current_indent_level)  File "/usr/lib/python3.7/json/encoder.py", line 431, in _iterencode    yield from _iterencode_dict(o, _current_indent_level)  File "/usr/lib/python3.7/json/encoder.py", line 405, in _iterencode_dict    yield from chunks  File "/usr/lib/python3.7/json/encoder.py", line 438, in _iterencode    o = _default(o)  File "/home/pi/dev/sopare/sopare/numpyjsonencoder.py", line 38, in default    return json.JSONEncoder.default(self, obj)  File "/usr/lib/python3.7/json/encoder.py", line 179, in default    raise TypeError(f'Object of type {o.__class__.__name__} 'TypeError: Object of type bytes is not JSON serializable`
What's wrong with this code. I use python3 from raspberry Pi.

bluesailor45, 2021-12-08:
Hi,
I get the same error when running sopare with python3.7. As I have seen in the project description sopare is only tested in python2.7 environment. Try running sopare unser python2.7.
I hope sopare will be updated to run also under python 3.
Maybe we should open a issue ?!

bluesailor45, 2021-12-08:
Hi,
sopare should run under python3 - see issue: **sopare in python3**. After downloading/cloning new project see also new dependencies described in readme file in spare-testing directory.
However: I have also installed this new files and installed the dependencies, but still got errors and was not able to run sopare (I am still checking why) - In another raspberry environment I was able to run sopare under python3 however. I will check what are the differences. If you are successful, please comment.

bishoph, 2021-12-10:
When you get errors you should post the error messages. I got several positive feedback that SOPARE runs with Python 3 incl. from myself :)

bluesailor45, 2021-12-10:
Hi Martin,

I am still testing. Currently I am facing issues with a new Raspbian Installation (Bullseye) which uses Python 3.9 as default. My issues are not associated with sopare (as far as I can see), but with the support of my mini USB microphone. I get errors from ALSA lib that the type in the configuration is not correct. As soon as I have overcome this issue, I am able to complete my tests with sopare on Python 3+. I will then give you feedback.

Thank you very much for this super product.

Very kind regards 
Kurt Feusi

bluesailor45, 2021-12-17:
Hi,
when running sopare.py (1.5.6) under python3.7 I get the following:
sopare is recording wav file, but then it closes with error:
Process worker for filtered data:
Traceback (most recent call last):
  File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/home/pi/sopare/sopare/worker.py", line 125, in run
    self.util.store_raw_dict_entry(self.cfg.getoption('cmdlopt', 'dict'), self.raw_character)
  File "/home/pi/sopare/sopare/util.py", line 108, in store_raw_dict_entry
    json.dump(json_obj, json_file, cls=sopare.numpyjsonencoder.numpyjsonencoder)
  File "/usr/lib/python3.7/json/__init__.py", line 179, in dump
    for chunk in iterable:
  File "/usr/lib/python3.7/json/encoder.py", line 431, in _iterencode
    yield from _iterencode_dict(o, _current_indent_level)
  File "/usr/lib/python3.7/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/usr/lib/python3.7/json/encoder.py", line 325, in _iterencode_list
    yield from chunks
  File "/usr/lib/python3.7/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/usr/lib/python3.7/json/encoder.py", line 439, in _iterencode
    yield from _iterencode(o, _current_indent_level)
  File "/usr/lib/python3.7/json/encoder.py", line 431, in _iterencode
    yield from _iterencode_dict(o, _current_indent_level)
  File "/usr/lib/python3.7/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/usr/lib/python3.7/json/encoder.py", line 438, in _iterencode
    o = _default(o)
  File "/home/pi/sopare/sopare/numpyjsonencoder.py", line 38, in default
    return json.JSONEncoder(self, obj)
_**TypeError: __init__() takes 1 positional argument but 3 were given**_
INFO:sopare.processing:stop append mode because of silence
INFO:sopare.buffering:stop buffering
INFO:sopare.buffering:terminating queue runner
INFO:sopare.recorder:Buffering not alive, stop recording

When running under python2.7 the following:

Capturing wav file runs OK, but when trying o create dictionary I get error: (see following)
........
.13777675301913467, 0.15378435347814873, 0.14845224802850057, 0.14580411949117664, 0.13977172468490975, 0.07205208052782061], 'token_peaks': []}
DEBUG:sopare.worker:meta = [{'token': 'stop'}]
INFO:sopare.recorder:Buffering not alive, stop recording
INFO:sopare.recorder:stop endless recording

_pi@raspberrypi:~/sopare $ python2 sopare.py -c_
sopare 1.5.6
recreating dictionary from raw input files...
Traceback (most recent call last):
  File "sopare.py", line 187, in <module>
    main(sys.argv[1:])
  File "sopare.py", line 100, in main
    recreate_dict(debug, cfg)
  File "sopare.py", line 130, in recreate_dict
    utilities.recreate_dict_from_raw_files()
  File "/home/pi/sopare/sopare/util.py", line 195, in recreate_dict_from_raw_files
    self.writeDICT(self.getCompiledDict())
  File "/home/pi/sopare/sopare/util.py", line 156, in getCompiledDict
    json_obj = json.load(raw_json_file, object_hook=sopare.numpyjsonencoder.numpyjsonhook)
  File "/usr/lib/python2.7/json/__init__.py", line 291, in load
    **kw)
  File "/usr/lib/python2.7/json/__init__.py", line 352, in loads
    return cls(encoding=encoding, **kw).decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 382, in raw_decode
    raise ValueError("No JSON object could be decoded")
**ValueError: No JSON object could be decoded**

I am at a loss!

Thank you for help.

bishoph, 2021-12-17:
A user reported (the same?) issue and made some fixes:
https://github.com/cahein/sopare/commit/58601158e465c8bec67e3b9bb07ccdc34fc6266b

Not sure if this solves the issue but it's worth a test. 

To verify and fix this issue one need to be able to reproduce it.  Please provide a step by step description and some background information about used OS/versions etc. (there are at least two Python versions in the comments).

bluesailor45, 2021-12-19:
Hi Martin,
thank you very much for the tip. This fix is really working and made the code with python3.7 working.
Thanks a lot for this great product!

bishoph, 2021-12-19:
You are welcome. Good to know that the fix is working. I need to install and setup a clean system to be able to verify and fix the issue without breaking Python 2 support ... maybe when and if there is some time between the holidays.