bishoph/sopare #96: diverse errors after install and running sopare.py -u

bluesailor45, 2021-11-23:
after having installed sopare on a rasperry pi 4 as suggested, I get the following errors: (which mostly could be corrected as described below). The following problem however could not be resolved by me:

when running in the sopare directory: "python sopare.py -u"

the following error appears:

  **File "/home/pi/sopare/sopare/filter.py", line 82, in n_shift
    self.data_shift = [ v for v in range(0, self.cfg.getintoption('stream', 'CHUNKS')/2) ]
  TypeError: 'float' object cannot be interpreted as an integer**

how can this error be resolved?

Now the errors which could be resolved by me:

1) error at python ./sopare/sopare.py -u
  -> module ConfigParser not found
   ==> change ConfigParser in /sopare/config.py from ConfigParser to "import configparser as ConfigParser"

2) Error sciPy module not found
   ==> install sciPy with: "sudo apt install python3-scipy" 

3) No module named pyaudio
   ==> install pyaudio with "pip install PyAudio"

4) syntax error in analyse.py line 198: try ... except ImportError, err: 
   ==> try ... except ImportError as err:

5) in file unit_tests.py line20: -> no module  named: test_analyze 
   ==> import test.test_analyze as test_analyze
   ==> import test.test_filter as test_filter

6) in module processing.py line 22:
   ==> import sopare.prepare

7) in module prepare.py line 21:
   ==> import sopare.filter

8) in sopare/plugin/print __init__.py line 23
   ==> print(readable_results)

9) in test_filter.py line 41
   ==> from : "        for x in xrange(0, len(data_object_array), self.CHUNKS):" 
   ==> to :   "        for x in range(0, len(data_object_array), self.CHUNKS): "

bishoph, 2021-11-23:
What branch are you using? I guess lots of this stuff (maybe all) has been fixed in the testing branch where you also get the installation steps in regards to python 3...

bluesailor45, 2021-11-23:
Hi,
I think I have installed it on the testing branch with the following github cli code:
"gh repo clone bishoph/sopare"

bishoph, 2021-11-23:
You clone a repository including all branches. You have to switch to a certain branch. Latest version in testing branch is 1.5.6 so if you are using a lower version you probably using the main brach which has no Python 3 support.

bluesailor45, 2021-11-23:
OK, thank you. Yes, when checking the code locally I think I downloaded the main branch. As I am not used to github, how do I switch to the latest testing branch?

bishoph, 2021-11-23:
If you are using git then this should be helpful:

http://rogerdudler.github.io/git-guide/

bishoph, 2021-11-23:
Closing issue

closed 2021-11-23 by bishoph

bluesailor45, 2021-11-23:
Thank you very much, Martin.
I managed now to clone the testing branch and it is working now very smoothly.

Thanks a lot.

Very kind regards 
Kurt