bishoph/sopare #98: testing branch python3 errors

cahein, 2021-12-10:
There is a problem left to fix in branch testing, resulting from upgrade changes in base64. Made minimal fixes in fork cahein/sopare. Didn't test it with python2. For python2 it would likely require an isinstance test of the result from base64.b64encode(obj_data).

bishoph, 2021-12-17:
Thx for the fix.

Can you please provide some more information about used OS/versions etc. as with my versions everything seems to run just fine. Also some kind of step by step how to reproduce description.

cahein, 2021-12-18:
Thanks for writing SoPaRe!

Running on debian bullseye. The error came up, when working with the dict.json.
base64.b64encode has a different return type in python2 (string) and python3 (bytes).

ptyhon3.8/library/base64.html
 base64.b64encode(s, altchars=None)
    Encode the bytes-like object s using Base64 and return the encoded bytes.

python2.7/library/base64.html
base64.b64encode(s[, altchars])
    The encoded string is returned.

bishoph, 2021-12-18:
Thanks for the clarification. Ok, I need to install and setup a clean new system. Maybe I get this done between the upcoming holidays.