bishoph/sopare #101: Please do not let that project die...

reparator, 2022-12-08:
Hallo,
komme erst in den letzten Tagen zu diesem Projekt und finde es grossartig!
Ich glaube auch, dass es für einen Insider nicht schwer sein kann die Adaption an Python3 zu machen. Vieles ist ja schon gemacht.
Beim Speichern/Verarbeiten der Testdateien wirft Python folgenden Fehler:
```
$ ./sopare.py -v -t test
...
...
Process worker for filtered data:
Traceback (most recent call last):
  File "/usr/lib/python3.9/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/home/pi/dev/sopare/sopare/worker.py", line 125, in run
    self.util.store_raw_dict_entry(self.cfg.getoption('cmdlopt', 'dict'), self.raw_character)
  File "/home/pi/dev/sopare/sopare/util.py", line 108, in store_raw_dict_entry
    json.dump(json_obj, json_file, cls=sopare.numpyjsonencoder.numpyjsonencoder)
  File "/usr/lib/python3.9/json/__init__.py", line 179, in dump
    for chunk in iterable:
  File "/usr/lib/python3.9/json/encoder.py", line 431, in _iterencode
    yield from _iterencode_dict(o, _current_indent_level)
  File "/usr/lib/python3.9/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/usr/lib/python3.9/json/encoder.py", line 325, in _iterencode_list
    yield from chunks
  File "/usr/lib/python3.9/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/usr/lib/python3.9/json/encoder.py", line 439, in _iterencode
    yield from _iterencode(o, _current_indent_level)
  File "/usr/lib/python3.9/json/encoder.py", line 431, in _iterencode
    yield from _iterencode_dict(o, _current_indent_level)
  File "/usr/lib/python3.9/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/usr/lib/python3.9/json/encoder.py", line 438, in _iterencode
    o = _default(o)
  File "/home/pi/dev/sopare/sopare/numpyjsonencoder.py", line 38, in default
    return json.JSONEncoder(self, obj)
TypeError: __init__() takes 1 positional argument but 3 were given
INFO:sopare.processing:stop append mode because of silence
INFO:sopare.buffering:stop buffering
INFO:sopare.buffering:terminating queue runner
INFO:sopare.recorder:Buffering not alive, stop recording
INFO:sopare.recorder:stop endless recording
```
Das gespeicherte File *.raw ist wohl leer (58 Byte):
```
{"id": "test", "characteristic": [{"fft": {"__ndarray__":
``` 
Der Dictonary bleibt leer weil das samplefile leer ist.

Please help!
Stefan

bishoph, 2022-12-09:
Hi Stefan,
there is a fix for this issue: https://github.com/cahein/sopare/commit/58601158e465c8bec67e3b9bb07ccdc34fc6266b
As I need to setup a complete new environment and have currently other priorities on my plate an "official" fix will not come soon...

Have fun.

reparator, 2022-12-09:
Thanks Markus,
with this commit it works even under bullseye. Recognition is quite good. 
Thanks a lot for your work and your very good documentation.
Regards
Stefan

closed 2022-12-09 by reparator